windelbouwman/ppci #1: Redesign project logo

windelbouwman, 2018-06-25:
At this point the logo is pretty minimalistic. We may require a new project logo.

pfalcon, 2020-01-03:
Oh, so those zeros and ones with py "cross" in background was actually a project logo? Never noticed ;-).

I usually consider this logo stuff waste of time, but this concept is too cute to pass by.

[image]
 
I guess no less important is a tagline/motto. The cool one, "LLVM, but worse", is [already taken](https://github.com/LLVM-but-worse), so, there's no other choice than to take the remaining "Like LLVM, but better". :rofl:

pfalcon, 2020-11-13:
> Oh, so

I assume it didn't work out with that logo, so I'm withdrawing that proposal.