windelbouwman/ppci #4: WebAssembly as backend

piranna, 2019-06-23:
It's interesting WebAssembly is used as frontend, but could it be added support to use it as backend too?

windelbouwman, 2019-06-23:
Thats actually implemented, see https://ppci.readthedocs.io/en/latest/reference/cli.html#cmdoption-ppci-cc-wasm

piranna, 2019-06-23:
Great! Maybe should be added to the readme file? :-)

windelbouwman, 2019-06-23:
This is a good idea! Thanks for the feedback!

piranna, 2019-06-23:
You are welcome :-)

pfalcon, 2019-08-04:
>> Great! Maybe should be added to the readme file? :-)
> This is a good idea! Thanks for the feedback!

It seems this was addressed by https://github.com/windelbouwman/ppci-mirror/commit/cd692923282561219fc4cff994a1bd3fdde4cb6d#diff-4fca3e8b2a5097bd09e6e331b570ade8 , so hopefully this ticket can be closed.

But looking at that commit raises other concerns, let me try to share them...

windelbouwman, 2020-01-12:
I'll close this issue, if needed, please re-open.

closed 2020-01-12 by windelbouwman