windelbouwman/ppci #6: Patches using git/github are accepted?

pfalcon, 2019-08-04:
While reading https://ppci.readthedocs.io/en/latest/support.html#how-to-submit-a-patch , I noticed that project requires submitting patches using Mercurial, and following special conventions, like:

> Create a hg bookmark, not a branch, and make your change.

(In majority of VCSes, a work would be done on a branch.)

Are contributions via this Github project accepted? It would be nice to clarify this matter.

windelbouwman, 2019-08-04:
The problem with mercurial is that the terminology is a bit different than most other systems. Branches are a long lived idea, and stick in the repository forever. Bookmarks are more equivalent to git's branches, they vanish when the branch is deleted.

I would rather keep the github a mirror, and work on mercurial as a base, but this is not set in stone, as a lot of people use git nowadays.

pfalcon, 2019-08-04:
> I would rather keep the github a mirror, and work on mercurial as a base, but this is not set in stone, as a lot of people use git nowadays.

I very well understand affinity to the existing project choices, but as you mention, a lot of people use git, and even github nowadays. And github may be as temporary as anything else before it, but it's something which allows to gather critical mass for some projects in some cases, something which was very hard before. So, as I always say, if github is just temporary, it makes sense to use it now as much as possible trying to get as much bootstrap out of it as possible.

So, thanks for not writing it off right away. Let's see if it might work out, https://github.com/windelbouwman/ppci-mirror/pull/8.

pfalcon, 2019-08-31:
> I would rather keep the github a mirror, and work on mercurial as a base

Here's what spam brought some time ago:

https://bitbucket.org/blog/sunsetting-mercurial-support-in-bitbucket

> After much consideration, we’ve decided to remove Mercurial support from Bitbucket Cloud and its API. Mercurial features and repositories will be officially removed from Bitbucket and its API on June 1, 2020.

Stupid unstable world...

windelbouwman, 2019-09-01:
Yes, I got the same spam, thanks for the notice anyway, it's time to figure alternatives.

pfalcon, 2020-05-31:
Recent several months showed that development process on github works quite well. The only remaining concern was that github project was named "ppci-mirror", but that was handled in https://github.com/windelbouwman/ppci/issues/64 . So, closing.

closed 2020-05-31 by pfalcon