windelbouwman/ppci #7: Clarification on the license choice

pfalcon, 2019-08-04:
I apologize in advance if this matter was well discussed already. It might make sense to link/summarize such a discussion from an accessible place then. In the meantime, license doesn't seem to be mentioned in the docs at all: https://ppci.readthedocs.io/en/latest/search.html?q=license 

So, what are the ideas behind choosing the BSD license for the project? Who are the expected contributors to the project who would enjoy such a license? Who are the users?

Thanks.

windelbouwman, 2019-08-04:
There is not really an idea behind the license. I just picked a liberal appearing one. Do you have strong opinions on a good license? I would like the software be usable wherever people want, including commercial products (allthough I would not recommend that :P).

pfalcon, 2019-08-04:
I hope to write a more detailed response. But as a prelude, just a  quote from random browsing the githubs:

https://github.com/digitalbazaar/forge/blob/master/LICENSE

> We are attempting to strike a balance between helping
the development community while not being taken advantage of by lucrative
commercial entities for our efforts.

We're definitely on a peak of usage of (pseudo)liberal licenses and ignoring and ignorance of the Free Software principles. But disillusionment is coming, and it already hits various parties (the above, the Commons Clause, etc.)

pfalcon, 2019-08-31:
Ok, wouldn't like to leave this without further arguments, at least for whoever may be looking later whether this topic was discussed...

> I would like the software be usable wherever people want, including commercial products (allthough I would not recommend that :P).

Nobody is going to use PPCI in a commercial product, because it's commercially infeasible. It's a) underdeveloped, undocumented, [not following best practices](https://github.com/windelbouwman/ppci-mirror/issues/5), etc. Commercial entities ain't no fools to bet on such projects if much better alternatives exists, e.g. LLVM. They would use it when it offers tangible benefits over alternatives like LLVM, e.g. b) being able to develop a backend or frontend for some hardware/language much easier than for those alternatives, and for that there should be a real documentation, real optimizations, real testsuite without commits like cfab16b86670c7ed9f64637ca06a21a4cb855d61 ("or disable until resolved" part), etc.

So, the question is who would bring the project from the state a) above to the state b), and why they would do that, given that the end result will be commercial vendors forking the project, withholding fixes, withholding improvements, withholding new targets? The only reason might be "to benefit community". But community benefit needs protection.

Nowadays, more and more parties notice (e.g. https://commonsclause.com/ , https://www.zdnet.com/article/open-source-licensing-war-commons-clause/ ) that unconditionally-welcoming-commercial-use-no-matter-what isn't really helping to maintain open source software, doesn't necessarily helps maintainers, and has only mediated benefit to the community (free advertisement, whoa).

But to say that the problem new is misguiding. In 1989, the BSD, MIT, etc. licenses were well existing, yet Stallman guy went to create GPL, which put community interests ahead of commercial entities', at least at the part of proprietarization of source code.

At this current random date and time GPL is very unfashionable (thanks to PR from those commercial entities, after all, who wouldn't want that somebody worked for them for free). But it very well exists and used by people who are not random passers-by in the field. A random focused example, given that we both in the "create your own Python" game: https://keithp.com/snek/ .

Note that this is not a call to switch to GPL - who am I to make such a call? My interest is pragmatic - I'd like to know who, when, and why would work on this project to move it further. And the "why" question is the most important, controlling the "who" and "when".

pfalcon, 2019-08-31:
(Ah, and as a full disclosure, I have a similar project - https://github.com/pfalcon/ScratchABlock . It's marketed as "decompilation framework in Python", but the direction isn't important - fully general framework should be able to perform program transformation in any direction. And no single minute I hesitated that the license for such a project should be GPLv3.)

windelbouwman, 2019-09-01:
Thank you for your detailed argumentation, this is insightful! I'm inclined to switching to GPL, but I'll give it some time. I'm curious to other opinions as well, but your reasons make sense. I just did not give too much thought about license choice. For me this is merely exploded hobby work, and at some point I figured I should pick some license.

What is the proper way of switching a license? Should it be included in the next release?

pfalcon, 2019-09-02:
>  I'm inclined to switching to GPL, but I'll give it some time.

Thanks for not rejecting the idea right away. And as PPCI is a multi-year codebase, decisions like that definitely should not be hasted, and should be thought out well, in sufficient time.

Nor, as I told, my aim was to call for GPL relicensing, just to clarify reasons for choosing the current licensing, and give an arguments why GPL might bve a good license for such a project. But in any case, license change should be driven by contribution process. E.g. someone saying "I'd love to contribute considerable improvements, but concerned about non-copyleft licensing". Or vice-versa, someone saying that they won't be able to contribute as much if licensing wasn't liberal.

> What is the proper way of switching a license? Should it be included in the next release?

I guess for the foreseeable future it should be just open for discussion. And existing contributors would surely have a first say here. So, ping @almarklein. (I guess there were more contributors, but that's who can be easily seen via https://github.com/windelbouwman/ppci-mirror/graphs/contributors . The rest probably don't have github account, or didn't use email address in commits matching one).

Otherwise, formally switching license from BSD to GPL is easy, again due to the fact that BSD is so liberal license that it doesn't put any restrictions on which other licenses BSD-licensed code may be combined with.

pfalcon, 2019-09-02:
(I also probably should elaborate my interest. I always thought that the project like PPCI should exist, but never was crazy enough to actually start one (at most, munging with partial functionality like aforementioned https://github.com/pfalcon/ScratchABlock). Now that it exists it would be a shame to not help it grow (or maybe just with cleanups). Unfortunately, I'm hands-full with the other projects, so can't be sure if and how I'm able to contribute. On the other hand, it's very easy to get sucked into such a cool project, and only years later to discover that where it will have gone is against aspirations of being involved originally. So, I'm sorry for possibly making too much forward-looking noise, but experience shows that any silent assumptions about such matters now leads to misunderstandings later.)

pfalcon, 2019-09-02:
>> For me this is merely exploded hobby work

> I always thought that the project like PPCI should exist

Btw, I kinda make an assumption that PPCI is something more than a "hobby work". For example, I may imagine including PPCI on godbolt.org took quite some effort, and wouldn't happen without a desire to expose PPCI to much wider audiences.

almarklein, 2019-09-02:
I've always been a fan of permissive licenses. I'm pretty sure that Python would be nowhere near where it is now if the ecosystem was not build with BSD-ish licenses. That said, I acknowledge that these licenses do not prevent open source developers being "used" by big corporations.

I have an aversion for GPL since it separates open source from commerce, building a wall between these two worlds, which cannot be a solution in my view. For some projects GPL might make sense, sure. Maybe it does for PPCI ... though I would be reluctant to contribute, because I cannot copy any of my code and re-use it in another project (unless that's GPL too).

I know that it does not happen on a large scale, but *some* companies do (want to) contribute to open source. We want this to grow and find other ways in which companies can play a significant role in financing development of OS. Like e.g. TideLift is trying (though I'm a bit skeptical since its VC funded), or e.g. [boss](https://www.boss.dev/). Cultivating this won't work with GPL. It will with BSD/MIT. If there is a license that still allows these things, but protects the community better, that would be great. I'm no lawyer and have trouble getting a feel how e.g. the commonsclause would work out.

pfalcon, 2019-09-07:
@almarklein: Thanks for reply, much appreciated! Let me go thru it and add some comments. Not for argument's sake, as I mentioned, it's not my intention to get the project converted to GPL (out of blue). If anything, I might try to show that, at this point of time, we (as a wide IT-related community) have a considerable perception skew/bias in the question of BSD vs GPL (or in more general, liberal vs copyleft) licenses.

> I'm pretty sure that Python would be nowhere near where it is now if the ecosystem was not build with BSD-ish licenses.

Well, it's pretty hard to assess that statement. It's a matter of fact that the evolution is such that all currently popular high-level languages are permissively licensed. It puts pressure on any new starter. But Python isn't new starter. I'd still think it's [survivor bias](https://en.wikipedia.org/wiki/Survivorship_bias). If you look at how many language project died, it would be the same 99.9% for both BSD/GPL. Anyway, if you want counter-example, it's GCC (and it survived and florished because it was commercially backed; but one can argue it remained what it is - a top quality open source compiler - thanks to GPL). 

> That said, I acknowledge that these licenses do not prevent open source developers being "used" by big corporations.

Fairly speaking, GPL doesn't prevent either. It does set up more elaborate rules, with the idea that if parties agree to follow them, they're more aligned in their values and approaches. And that's mostly what we can shoot for (and what I'm talking about). There're still enough room for abuse.

pfalcon, 2019-09-07:
> I have an aversion for GPL since it separates open source from commerce, building a wall between these two worlds

Sorry, but why and how? That's definitely not true. Linux and GCC are two obvious examples. Gazilllion of companies make money on Linux, and GCC supports all those targets because companies behind those targets went and pay for them to be implemented. Moreover, dual-licence GPL-vs-commercial was a default model for new startups and small companies some ten years ago, and still practiced widely enough. E.g. [wolfssl](https://www.wolfssl.com/license/) is GPL+commercial. This guy: https://github.com/cpq , when got bored with working at Google and went for his startup, cast all of his liberally licensed projects to GPL+commercial, from more or less well-known [Mongoose](https://github.com/cesanta/mongoose) web server (the original MIT-licensed version forked as https://github.com/civetweb/civetweb) to smallest pieces like this this regex lib: https://github.com/cesanta/slre (original version: http://slre.sourceforge.net/).

But not only small companies love GPL. MySQL was GPL+commercial when it was a small company. When Oracle bought it, it's still GPL+commercial. But Oracle bought not only MySQL. They bought BerkeleyDB and cast it not just plain GPL, but Affero GPL.

So, commerce lllloves GPL. That part, which sells (original (oh, or acquired :-) )) software at least. There're other commerces too. E.g., ones which sell us ~~advertisement~~ services. They just want to mix and match whatever is available outside in different shapes and sizes, and of course, GPL with it's "wait, there're rules to follow" attitude is just a nuisance to them. Liberal licenses which literally respond to GPL's claim above "oh, f^%& off", is much better match for them. And as they're deep in advertisement (can't sell much of services nowadays, can mostly sell advertisement), they tell us they don't want GPL.

Then there're startups in stealth mode. If they use GPL now, they won't be able to relicense as GPL+commercial later, so they say BSD. That's very small fraction of stakeholders, but hey, everyone want to be reach and famous. So we're telling to ourselves "Hey, tomorrow I'll stand from an armchair, pick up all that stuff around, and build a spaceship. That'll work only with BSD, so yeah, I want BSD!"

That's how we ended up where we are.

pfalcon, 2019-09-07:
> though I would be reluctant to contribute, because I cannot copy any of my code and re-use it in another project (unless that's GPL too).

Of course you can use *your* code, GPL is non-exclusive license, as a copyright holder, you can do whatever you want *in addition* to what GPL allows/requires. That's the basis of tested and tried "GPL+commercial" strategy described above. But you cannot just copy and reuse *somebody else's* code. But that's kinda the whole point: GPL sets up a playfield and process where different parties need to be aware and respect rights and needs of other parties, instead of just cheerfully plucking results of their work.

> I know that it does not happen on a large scale, but some companies do (want to) contribute to open source. 

Here I'm not sure I agree. Open-source has long become a basis of modern IT, a lot of companies contribute it. At least, it shouldn't be a too big a problem finding an open-source job. And that's how companies contribute to Open Source - they higher people to work on it. But there's one important point - this happens on *their* terms. Expecting that if you've got an open-source project and somebody will pay you for that is, well, ungrounded. Companies pay for Open Source not because it's Open Source, but because it provides value needed to them. And there's harsh competition in the open source ecosystem, as anywhere else.

> Like e.g. TideLift is trying (though I'm a bit skeptical since its VC funded), or e.g. [boss](https://www.boss.dev/).

I'm not sure if you actually looked by the link you gave, but there're 2 bounties there: one a test one, and another for $10: https://www.boss.dev/issues/open . I'm watching sites like that since [cosource.com](http://www.searls.com/linuxforsuits/cosource_interview.html) times, and may report that sites like that pop up and go down like bubbles (indeed, it's hard nowadays to even google up references of aforementioned cosource.com).

It's nice unicorn dream that somebody will come and pay for your open source, but in all fairness, why pay for something which is already free? Any business-oriented person would quickly come up with an answer: because some very small number, exceptional specimens of it bring actual value. That's why GCC was funded. Going back to the PPCI case, who will pay for something being added to it, if money has been already paid to have stuff added to GCC and LLVM, or can be paid to those projects (the industry leaders)?

pfalcon, 2019-09-07:
> companies can play a significant role in financing development of OS. Like e.g. [] Cultivating this won't work with GPL. It will with BSD/MIT. 

I'm sure "OS" is typo there, but it's actually a good thinko. Linux is the OS where heterogeneous (no central authority) company funding finances the development. Yearly budget is definitely tens of millions. And some say such model and ecosystem was only possible thanks to GPL.

And in the end, if you want to donate money to project, or sponsor development, you would do that (and commercial entities have done that) regardless of license, GPL for sure doesn't have any terms against that.

By the end, I'd like to provide thesis-like summary:

1. There's seems to "cinderella story" in (individual) open-source community that such reach dude may come by and give (loads of) money, and that it may happen only with BSD (or similar liberal) license. This may happen only under exceptional conditions. Life is short, can sit and wait for that to happen whole life.
2. It's common saying that for something exceptional to happen you should be at right place at right time. We're in the era of commoditization of open-source software, so that's not enough, you should be in right shape too. Or otherwise there're a bunch of competitors just around the corner to choose from. In PPCI case, it's GCC and LLVM, so why would PPCI be chosen?
3. It's era of commoditization of open-source software again. Housewifes won't pool up and donate to PPCI, they write open-source software themselves. And they don't need a compiler, they have GCC, remember? (Advanced even have Clang).
4. Generally, money is not the most important matter in life. And desire to get paid for each and every hobby you have may cost you quite a bunch of both money and hassle in life (approach of "sit and wait" discussed above, a more realistic one is: quit job, move to another place to get a job for your hobby, get interested in sth else, quit, move, repeat. Sounds awful ;-) ).
5. The talk is again whether other license may set a better playfield for contribution to PPCI. I'm not sure it will. I'm not sure it will make a difference, even if it could - PPCI is such a special project, that maybe never enough people will contribute to it at all. I just know that BSD is a good license for grabs, and not too good license to work on in your own free time.

almarklein, 2019-09-09:
> >  though I would be reluctant to contribute, because I cannot copy any of my code and re-use it in another project (unless that's GPL too).
>
> Of course you can use your code, GPL is non-exclusive license, as a copyright holder ...

You're right here, I stand corrected.

> >  I have an aversion for GPL since it separates open source from commerce, building a wall between these two worlds
> 
> Sorry, but why and how?

Because any code that uses *anything* that is licensed GPL, is forced to be released under the GPL. In my view this makes it much harder to have a heterogeneous ecosystem in which we have both free software and companies building stuff using this software.

I understand that one can adopt a dual license to allow commercial use, and surely this has worked fine for some projects in combination with large companies. But for smaller companies it's much harder. You either develop OSS for free, or are paid by a big corp to do it, if you're lucky.

> There's seems to "cinderella story" in (individual) open-source community that such reach dude may come by and give (loads of) money, 

That's not at all what I mean, or how I'd want things to work.

The "ideal world" that I've always felt that a community based on BSD licenses could *potentially* help support, is one in which makers produce free software and *also* become entrepreneur by building innovative things with the tools that they collaboratively create. The creators and people benefiting financially are part of the same group of people. I don't think such a creative community would be possible with restrictions like GPL imposes.

The problem in the above vision is that as companies grow they tend to lose interest in the community. More so, it's way too easy for big corporations to take advantage of the things being created without contributing back. That's something we should address, but in my view, GPL is not the answer.

pfalcon, 2019-09-10:
@almarklein: Thanks for discussion! I tried to pack many answers in my previous response, and don't have much to add. And I of course can very well see benefit in a project remaining BSD. Even sans "becoming entrepreneurs" clause, I can find a userstory where remaining BSD is rather helpful, e.g. "I'm working on a implementation language X, per the landscape it must be liberal license, and per modern standards, I want there be compiler/JIT compiler. Merging parts of PPCI into my project without any licensing baggage is very helpful." Except I don't think that PPCI is yet there to make that attractive.

So, once again, this is a discussion/clarification ticket, and not a direct call to change anything. I hope it can stay open for a while, to collect feedback from any interested parties. Perhaps even to correlate feedback and project progress, and maybe changes in attitudes of participants over time, if any.

Thanks!

almarklein, 2019-09-11:
Thank you too @pfalcon :)

> Re: Of course you can use your code, GPL is non-exclusive license, as a copyright holder ...

No, wait. If I contribute code to a GPL licensed project, I still "own" the code per copyright. However, if someone makes an improvement to the code that I contributed, things start to become complicated, because now it has "GPL-sauce" on it and it cannot be anything else than GPL :)

That said, I also realize that this is a problem *here* because of the ecosystem: I am biased towards BSD because the Python ecosystem has a strong bias towards that.

mmm ... a lot of food for thought ...

almarklein, 2019-09-11:
(sorry, this thread is way off-topic by now 😄)

I did some thinking, and this led me to the following thought experiment:

Let's imagine the Python ecosystem, but consisting mostly of copyleft (e.g. GPL) licensed projects. Now let's say that I am active in this community, and I have an idea for a product that I want to build a business with to support myself and my family. I would track all the dependencies that my product has. But how would I proceed from there? If all the dependencies offer a commercial license in addition to GPL, I could (hopefully) pay the fees and go for it. But for each dependency that is GPL-only, I have to find (or create) an alternative.

I'm genuinely interested in your thoughts on this @pfalcon. I *want* part of the income that my product generates to flow back to the people that created these projects. Actually, I would want to be *obligated* to do so. But if half my dependencies prohibit commercial use, grrr. Perhaps within such a community there could be a platform to make it so easy to setup a commercial license that almost each project has one  ...

pfalcon, 2019-09-11:
> However, if someone makes an improvement to the code that I contributed, things start to become complicated, because now it has "GPL-sauce" on it and it cannot be anything else than GPL :)

Sure, and that's the whole point, and they're not complicated, they're clear - GPL code doesn't "belong" to someone, it belongs to "community", with "community" defined as "any actual or potential user of code, past, present, or future". Or in other words, as soon as the matter becomes public (more than 1 person works on it), it remains public.

> I am biased towards BSD because the Python ecosystem has a strong bias towards that.

I don't know why you have that impression, I don't share it at all (I'm with Python since 1.5.2). At "golden" (or dark, depending how you see it) ages, there was a lot of GPL Python modules. It changed not specifically for Python, but with the ecosystem as a whole, with commoditization and inflation of open source.

almarklein, 2019-09-11:
> I don't know why you have that impression, I don't share it at all (I'm with Python since 1.5.2). At "golden" (or dark, depending how you see it) ages, there was a lot of GPL Python modules. 

Ok, then I am biased towards BSD because I joined the community at v2.5 and the Python ecosystem had already started moving towards BSD by then :)

I recall open sourcing one of my first projects, and picking a GPL license for it. Then I had a discussion with Fernando Perez about the license and he convinced me to change it to BSD (would be interesting to find that discussion back , but couldn't find it just yet).

pfalcon, 2019-09-11:
> Let's imagine the Python ecosystem, but consisting mostly of copyleft (e.g. GPL) licensed projects.

It's improbable situation. GPL was conceive to stand as a shiny island among the proprietary software. It remains a shiny island among permissively-licensed software, which is effectively a one step from proprietary. A shiny cathedral among the smelly bazaar ;-). All of software doesn't need to be GPL (nor it needs to be open-source).

> I have to find (or create) an alternative.

I'm not a pro in business plans. But layman-wise, I would assume it to be a solid foundation for business plan, that if you want to sell something, it should be something that you created ;-).

For more specific cases, there may be more specific means to deal with that. E.g., you can easily use GCC to build your proprietary software, and GPL doesn't interfere with that in any way.

And overall, we're talking whether for a project like PPCI, the GPL would be a better choice, not for GPL in general. It would be interesting to hear your business-case userstories which would be enabled by BSD and disabled GPL.

pfalcon, 2019-09-11:
> I recall open sourcing one of my first projects, and picking a GPL license for it. Then I had a discussion with Fernando Perez about the license and he convinced me to change it to BSD (would be interesting to find that discussion back , but couldn't find it just yet).

Better tell what the project it was, because again, there's no "the best license", each one serves its purpose and is used in specific circumstances.

(And yeah, we're going offtopic ;-) ).

almarklein, 2019-09-11:
> And overall, we're talking whether for a project like PPCI, the GPL would be a better choice, not for GPL in general.

Yeah, I'm talking about the more general case here, haha. 

> It would be interesting to hear your business-case userstories which would be enabled by BSD and disabled GPL.

I'm not sure if I correctly understand what you mean, but ... in the current (mostly BSD) Python ecosystem, I can have an idea, build something new based on the tools that I know, and turn it into a product (or SaaS, or whatever). Done! This lack of friction can be a boost for creativity. That's what I meant, anyway. In retrospect, perhaps some friction would be good if it helps prevent misusage by big corps.

By the way, I'm not claiming that it should be possible for any idea to be a product to generate money. Sometimes you just wanna have fun. But sometimes a project becomes more serious, and brings benefit to many others. In such a case it makes sense to be appropriately rewarded.

> Better tell what the project it was, ...

It must have been either Pyzo (a Python IDE) or Visvis (a visualization lib). The argumentation would certainly make sense for the latter, but I think it was about the former. Still can't find it, probably lost in an issue on Google Code, but I did find [this](http://nipy.org/nipy/faq/johns_bsd_pitch.html) from around the same time. It's a plea from John Hunter for package authors to adopt BSD instead of GPL.

I still believe that the community leaning towards BSD has been instrumental to the growth of the (scientific) Python ecosystem. But we also fell into a trap. The following quote may have been true then, but no longer applies today (at all): *"In my experience, the benefits of collaborating with the private sector are real, whereas the fear that some private company will “steal” your product and sell it in a proprietary application leaving you with nothing is not."*

> ... there's no "the best license", each one serves its purpose and is used in specific circumstances.

I agree to some extend, but the ecosystem matters as well. A project using GPL in a BSD-oriented ecosystem may have trouble getting adoption. So it appears that BSD also has a viral nature, when looking at it from a community perspective ;)