windelbouwman/ppci #13: Elaborate on what's "c3"

pfalcon, 2019-09-02:
I'm still not 100% sure, but it seems that "c3" referred in the README is an adhoc language developed specifically for/with PPCI. This fact should be made clear in README, and elaborated in the docs, where it's still not exactly clear: https://ppci.readthedocs.io/en/latest/reference/lang/c3.html

And for reference, see what google finds for "c3 language":

* https://marketplace.visualstudio.com/items?itemName=rsamp.language-c3
* https://www.reddit.com/r/ProgrammingLanguages/comments/cdx2co/the_c3_programming_language_draft_design/

What is probability that those "c3" is the same "c3" as used in PPCI?

windelbouwman, 2020-01-02:
c3 is actually the root cause for this project. I thought, C sucks in some ways, I can do better, and here we are, writing a compiler for C (irony, I love it :)). Anyways, it is an example language showing the use of the intermediate code, it is simpler than C, but also a NIH language. In any case it's unrelated to the google hits you found.

pfalcon, 2020-01-03:
> c3 is actually the root cause for this project.

Ok, thanks for explaining that, and it's the reason why I don't rush to submit changes which I think would be useful, and instead first try to ask questions and raise discussions first.

Anyway, my point is that a newcomer should get enough hint of what c3 is just from README, and at the same not be scared that they are driven into NIH kingdom (i.e. idea which they should get is that c3 exists and they *can* use it, not that they *have* to use it to play with ppci).

So, no let me try to find a compromise between those ideas and the fact that c3 is your dear pet project ;-). I'll post PR shortly.

pfalcon, 2020-01-07:
Ok, closing this, thanks!

closed 2020-01-07 by pfalcon