windelbouwman/ppci #16: registerallocator: Typo: "coalesc" instead of "coalesce"

pfalcon, 2019-09-02:
Docs for registerallocator module https://ppci.readthedocs.io/en/latest/reference/codegen/registerallocator.html have a typo in a few places, "coalesc". Worse, such a typo is also in a method name: https://ppci.readthedocs.io/en/latest/reference/codegen/registerallocator.html#ppci.codegen.registerallocator.GraphColoringRegisterAllocator.coalesc

Should be fixed nonetheless ;-).

pfalcon, 2019-09-03:
Another, and very repetitive typo: `Arithmatics` instead of `Arithmetics` (dozens of occurrences).

windelbouwman, 2019-12-31:
Agreed those typos must be fixed!

pfalcon, 2020-01-01:
Thanks for acking. Note that this tangentially touches https://github.com/windelbouwman/ppci-mirror/issues/28 - starting editing real code, like identifiers, one would really want to be sure update is consistent and doesn't bring regressions. Of course, I can run all the different test methods for now ;-).