windelbouwman/ppci #9: cc: Tighten up error messages

pfalcon, 2019-09-02:
~~~
2019-09-02 15:32:35,652 |    ERROR |       root | Who is this "printf"?
2019-09-02 15:32:35,652 |    ERROR |       root | (hello.c, 3, 5)
File : "hello.c"
    1:int main()
    2:{
    3:    printf("hello world\n");
          ^ Who is this "printf"?
~~~

Love the irony, but suggest to make error messages more formal.

pfalcon, 2019-09-02:
Oh, and btw, it's warning, not an error ;-).

windelbouwman, 2019-12-31:
What is your opinion about very descriptive error messages? Once I used rust, I noticed the compiler error messages are extremely detailed, where as gcc used to be pretty terse.

pfalcon, 2020-01-15:
Sorry, this was one of the smallish things I wanted to fix, but overlooked further discussion here...

> Once I used rust, I noticed the compiler error messages are extremely detailed, where as gcc used to be pretty terse.

They used to be terse, but due to noxious influence of clang, in recent gcc versions they're now bloated. Just kidding ;-).

> I would like to try out an experiment with super detailed error messages...

Well, so generally, I'm a big fun of self-documenting, self-describing, self-diagnosing software. But C, C is the foundation. It was created by god and just communicated thru its prophets K&R (who, being humans, made a few stupid mistakes writing down god's will). So super-detailed error messages? Isn't everyone already knows C by heart? ;-)

Anyway, on more serious account, I don't care is there're super detailed errors (well, I may be, if they will start to spam me, like 100 super-detailed error messages sounds horrible), as long as the start with telling the ground truth of what happens (leaving guesswork for next lines). In this regard, colloquialisms  like "Who is this "foo"?" aren't really helpful. If there's an *undeclared* identifier, compiler should just say that. In particular, it should not lie that identifier is *undefined*, because C doesn't require an identifier to be defined, only to be declared. (It's linker's task to make sure that each identifier was actually defined.)

So, my attempt at solving this is in https://github.com/windelbouwman/ppci-mirror/pull/46 . I personally would be fully satisfied by that patch.

But let me triage your proposed message:

> The name "printf" was not defined, this can have several causes:
> 1. you forgot a forward declaration (link to wikipedia here?). Note that in C symbols must be declared before they can be used.
> 2. you forgot to include a header file at the top of the file

So again, rules: a) tell the truth; b) do not lie. "not defined" is not truth, it's "not declared". Then it says "several causes" and lists exactly 2. Formal reading of that is that there's exactly 2 causes, which is of course not true. In general, there's usually 1 way to do something right, and potentially unlimited number of ways to do something wrong. So, truthy, not-lying message would be:

"The name "printf" was not declared, some possible causes:
1. It was misspelled.
...
"

> Note that in C symbols must be declared before they can be used.

This is **not** true. You can call a function without declaring it (modern compilers issue warning). Its prototype is then assumed to be `int func()`. Note that `func()` is exactly one of the stupid mistakes made by K&R writing down the god's will. I hope you know what `int func()` means - "function taking arbitrary arguments". What about `int func(...)`? Not valid syntax in C. So, how to declare a function taking 0 arguments? `int func(void)`. That's pretty big (and in my list, about the only) idiocy in C. The rest of syntax, me loves. "Lexer hack?" Piece of cake.

windelbouwman, 2020-01-16:
> I hope you know what `int func()` means - "function taking arbitrary arguments". What about `int func(...)`? Not valid syntax in C. So, how to declare a function taking 0 arguments? `int func(void)`.

No, I did not. :scream: Having programmed in C for the past seven years now, this comes as a surprise :)

Good other points about the error message, this should be correct and not lie, I fully agree. Maybe we can enable detailed errors with an option flag, so expert users can get the terse errors and novice users get the background story?

pfalcon, 2020-01-16:
> Maybe we can enable detailed errors with an option flag, so expert users can get the terse errors and novice users get the background story?

Well, point I tried to make that these detailed errors are not that easy to prepare, require some work, meaning "some work instead of another work". If you're interested in that, certainly, please go ahead. And if you really want to have an impact, these would need to be enabled by default. But then just more thought need to be put how to make UX good. E.g., maybe only first 10 errors to be detailed, and the rest switch to terse, one-line format, etc.

Btw, there's already some maintenance require for existing "smart" stuff, here's what I get for example:

~~~
2020-01-16 19:49:34,617 |    ERROR |       root | Undeclared identifier: "puts1"
2020-01-16 19:49:34,617 |    ERROR |       root | (hello.c, 8, 5)
File : "hello.c"
    6:int main()
    7:{
    8:    puts1("Hello, World!");
          ^ Undeclared identifier: "puts1"
    9:    /* As we don't have any C startup code in this example, and main is
   10:       the executable entrypoint, we can't just return, but have to
   11:       call exit syscall explicitly. */
"puts1" was not defined, did you mean "puts"?
"puts1" was not defined, did you mean "putc"?
"puts1" was not defined, did you mean "puts"?
"puts1" was not defined, did you mean "putc"?
~~~

As you can see, suggestions are output twice. I looked for 30s into that, but didn't see anything wrong on c/semantics.py side.