windelbouwman/ppci #17: Promoting the project

pfalcon, 2019-09-03:
As I mentioned in some other tickets, I was totally astonished to have missed the PPCI project for so long, despite actively watching scenes of both hmm, "hobby" compilers, and compilers-in-Python. Poor souls like me should be helped, there should be effort to spread the word about PPCI. This ticket is created to hopefully help to coordinate the effort.

pfalcon, 2019-09-03:
This is of course predicated on deciding what's the primary "home" of the project, e.g. issues mentioned in https://github.com/windelbouwman/ppci-mirror/issues/6 . Generally, if promotion is the aim, headquartering on Github is a natural choice, any other site lacks critical mass of folks around.

pfalcon, 2019-09-03:
Ok, my humble post to Reddit's /r/Compilers: https://old.reddit.com/r/Compilers/comments/cyo7et/pure_python_compiler_infrastructure_yes_easily/ brought the project 8 stars, which is still 30% improvement ;-).

pfalcon, 2019-09-03:
I also saw PPCI was posted on Hacker News: https://news.ycombinator.com/item?id=16847693 , but got ~zero ring: of 2 upvotes, one is mine. This again shows the importance of coordinating the effort: people in the know should help with initial upvotes, so it didn't drown in the noise right away, and other folks had a chance to spot it.

windelbouwman, 2019-09-03:
Nice job! You triggered some attention, thanks!

pfalcon, 2019-09-14:
PPCI is now included in the https://github.com/mahmoud/awesome-python-applications list (https://github.com/mahmoud/awesome-python-applications/issues/96).

windelbouwman, 2019-09-14:
I noticed! Thanks for doing this!

windelbouwman, 2019-09-14:
Did you already play with this script? https://github.com/windelbouwman/ppci-mirror/blob/master/tools/ppci_explorer.py

pfalcon, 2019-09-14:
> Did you already play with this script? https://github.com/windelbouwman/ppci-mirror/blob/master/tools/ppci_explorer.py

I saw it, but didn't pay much attention, now tried. What can I say, cute! ;-) Also should go in the README, so people could try it quickly.

pfalcon, 2020-01-01:
0.5.7 ann on Reddit: https://old.reddit.com/r/Compilers/comments/eihcuc/ppci_057_pure_python_compiler_infrastructure/