windelbouwman/ppci #21: README: Mention projects PPCI interoperates with

pfalcon, 2019-09-04:
It's pretty good point that PPCI tries to interoperate with other projects, like JVM, LLVM, CastXML, etc.

That should be a pretty good selling point - that PPCI is not just a NIH thing in itself, but actually interoperability platform. Should draw attention to that by explicitly listing projects in README.

windelbouwman, 2020-01-11:
Interoperability is not in very good shape at this point. Best interoperability might be the wasm part, so we can in and export wasm blobs, which is helpful. But it might be good to list at least the ideas and future directions, to set proper boundaries to this project.

pfalcon, 2020-01-11:
> Interoperability is not in very good shape at this point.

You must be kidding me ;-). Everything is relative, let's compare "interoperability story" with my cute project, https://github.com/pfalcon/ScratchABlock . Has own IR (well, inspired by some radare "plugins", with radare itself not exploring it well enough IMHO). Good news is that all IRs are essentially the same, with only syntax and level of adhockery different. So, do I provide converters to any of other formats I list in my IR zoo https://github.com/pfalcon/ScratchABlock/blob/master/docs/ir-why-not.md ? Nope. My list of useful 3rd-party tools is also rather short: https://github.com/pfalcon/ScratchABlock/blob/master/docs/tools-3rd-party.md . People coming and asking where's x86 loader, and I'm replying "x86 is the most boring arch possible, I'm not wasting my life on it, write the loader yourself".

So, that's a case when "interoperability is not in very good shape". Also explains why, having a similar program analysis project myself (which I intended to continue), I assess that it's more beneficial to contribute to PPCI at this time.

pfalcon, 2020-01-11:
So, my idea how this can be presented. Have a top-level section in the docs, with any of the bullet-points below, plus subset of the most important ones in the README:

* PPCI aspires to be command-line compatible with classical Unix "cc" (C compiler) and more specifically, GNU GCC and Binutils, so ultimately, PPCI tools are drop-in compatible with those.
* PPCI can be used with "make" or any other build tool.
* PPCI supports x86, x86_64, ARM, STM8, ... architectures.
* PPCI supports ELF, COFF PE, S-Record, ... formats for 
* PPCI supports Dwarf debugging information.
* PPCI support WebAssembly.
* PPCI can interoperate with LLVM IR.
* PPCI can process C, Pascal, Python source code.
* PPCI can import JVM and Ocaml bytecode.
* PPCI uses well-known human-readable and machine-processable formats like JSON and XML as its tools' formats.
...

I hope that gives an example of what "interoperability" list should be ;-). If PPCI has a tiniest bit of support for something - list it! Level of support is of course varies, but that's the whole points - show the people that PPCI can deal with all those stuff, and aspires to. Then let people contribute to improve support in the area they're interested in, because in know way a single human can address all that deep enough.

windelbouwman, 2020-01-11:
This list is a good idea! I'll update the readme with this ideas.