windelbouwman/ppci #26: RFC: Using rebase model for changes

pfalcon, 2019-09-07:
When using "Merge" button for pull requests in github web ui, the default setup is to always perform git merge operation. That means that even small single commit in a PR leads to 2 commits in the repo: the original commit and ~~ugly~~ useless "merge" commit.

Many projects instead use rebase-based model, where merge commits are never used, and new changes instead integrated into the main codebase in a seamless (merge-less) manner. Github UI supports that way either, it just should be set as the default in project settings, as describe here: https://help.github.com/en/articles/configuring-commit-rebasing-for-pull-requests (The recommended setup is that among 3 checkboxes, only "Allow rebase merging" is ticked).

@windelbouwman, I hope you would agree this is useful setup and can make this change.

windelbouwman, 2020-01-04:
Hi, I changed the setting in the github settings, I forgot to use it, but other than that this seems good to me. I lean more toward the merge strategy, since it preserves actual history (no history rewrite), but I have no super strong preference. I might decide from time to time to merge when the change is somewhat larger, but for smaller changes I will apply the rebase strategy.

pfalcon, 2020-01-04:
> I lean more toward the merge strategy, since it preserves actual history (no history rewrite), but I have no super strong preference. 

Thanks for the change! I would argue (actually I did in the original description) that it's merges "bend" the actual history, e.g. there was no "merge" commit in someone's original PR, but that merge commit appears in the mainline history. Then depending on the timing of the PR's commits and how much went into the mainline in the meantime, "merge" commit and actual commits may be quite distant. All that rather complicates tracking project history, why again many project with enough commit number and activity select the rebase model. With it, the situation is simple - the changes appear in the project history exactly at the point they were merged into the project, i.e. when they become a part of it. And then there's no need for "merge commit" placeholder, which is again, only adds noise.

I hope this model will serve PPCI well either.

closed 2020-01-04 by pfalcon