windelbouwman/ppci #27: Outdated package on PyPI: 0.5.6 (latest is 0.5.7)

pfalcon, 2019-09-14:
https://pypi.org/project/ppci/ :

> ppci 0.5.6

Would be nice to upload the latest version there.

pfalcon, 2019-09-14:
Actually, looking at the package release dates https://pypi.org/project/ppci/#history vs 93e955969a2c86fe2e87b329a6215fdcfa05c503, it could be that PyPI has 0.5.7, uploaded before the release version bump.

Still confusing situation and would be nice to fix it.

pfalcon, 2019-09-14:
Actually, maybe just release 0.5.8 ;-).

windelbouwman, 2019-12-31:
I uploaded the latest version to pypi. Can this issue be closed?

windelbouwman, 2019-12-31:
> Actually, looking at the package release dates https://pypi.org/project/ppci/#history vs [93e9559](https://github.com/windelbouwman/ppci-mirror/commit/93e955969a2c86fe2e87b329a6215fdcfa05c503), it could be that PyPI has 0.5.7, uploaded before the release version bump.

No, I just reserved a version number for future release. The next release number is already known in advance.

pfalcon, 2020-01-01:
Posted ann on /r/Compilers: https://old.reddit.com/r/Compilers/comments/eihcuc/ppci_057_pure_python_compiler_infrastructure/ . @windelbouwman, could you queue up sending an ann to python-announce mailing list: https://mail.python.org/mailman3/lists/python-announce-list.python.org/ ?

Closing otherwise, thanks for the release!

closed 2020-01-01 by pfalcon

windelbouwman, 2020-01-01:
@pfalcon I was unaware of the python announce list, what does it mean?

pfalcon, 2020-01-01:
@windelbouwman: Well, it's a list used to announce Python-related conferences, but much more frequently, to announce releases of Python software (as in: software in Python). It's just another channel to spread a word about the project. While community can help with posts on discussion sites like HN/Reddit, I'd think that making "official" announcement (and posting to python-announce seems as "official" as it can get ;-) ) is a prerogative of project maintainers.

You can get a sample of how people do it in the archives: https://mail.python.org/archives/list/python-announce-list@python.org/ . The only suggestion I'd personally have is to always start with (at least) a short description - many people fancy starting with "foo version X.Y has following changes: ...", and that's pretty boring to anyone who doesn't know what the heck "foo" is, and that's the majority of readers. Anyway, for first time, just copying a few paragraphs from README into email should work.

Thanks!

windelbouwman, 2020-01-01:
I has been posted: https://mail.python.org/archives/list/python-announce-list@python.org/thread/E2YQTF3PERM4OSR6UQ2NK2UDZD5RSKJO/