windelbouwman/ppci #29: Plans/future for PPCI?

pfalcon, 2019-12-30:
It's that time of the year which calls for sum up what was done for a year and lay further plans. I would  like to bring a question of what the future may hold for PPCI. I discovered this project only in August 2019, and was really astonished by its topic and scope - that's something I wished I would have started and done. And with that, it's just a project in pretty early stages re: any usability for "real world" usage. I quickly submitted a few patches, then started to watch the project to see where it goes. Well, over one month it didn't go anywhere, which is quite understood given summer/vacation time. I at least tried to dump whatever ideas of improvements I had in mind as tickets, so they aren't forgotten, and prepared for more patience.

Well, now by the end of the year, it's fair to say that beyond some activity in May-July, and my humble patches in Aug, totalling ~25 commits: https://github.com/windelbouwman/ppci-mirror/graphs/contributors?from=2019-01-01&to=2019-12-31&type=c , there was no much activity on the project. That's unlike pretty steady progress over previous 2.5 years.

So, @windelbouwman, as the author and owner of the project, I would like to ask you what plans/feelings you have regarding PPCI, and what people interested in it can expected. Thanks for your time and effort developing PPCI, and hopefully, to answer this question too.

windelbouwman, 2019-12-31:
Hi @pfalcon , it's indeed time to contemplate :).

Regarding ppci, I was a bit distracted by rustpython and lognplot projects I am playing with, so that's why there is no progress on ppci.

Features I would like to see in ppci:
- C frontend completion
- Other languages support (fortran and pascal in particular)

Part of the reason why I stopped working on this project was my doubt about rust. It's a great language and I was doubting if I should rewrite all of ppci in rust. The performance would be better, but it would be some work. I now think this should probably not happen. Having a compiler in python is very good, and I would like it to keep it that way.

My general aim would be to implement as much as possible into the ppci package, so it can really become a tool which can handle pretty much everything.

Another idea I have is reducing the dependencies upon the python stdlib, since I understood that some modules might be dropped, it might be handy to rely on as little builtin libraries as possible, say only struct, os for example.

Another near goal I have is compiling a sizable C project, just to test the C frontend. I tried libmad and the links browser as test projects. Compiling the linux kernel would be great as well :)

Do you have any specific directions you would like to see the project move?

pfalcon, 2019-12-31:
I appreciate detailed and prompt reply!

> I was a bit distracted by rustpython and lognplot projects

That's what I saw myself too. I actually wondered in lognplot is somehow related to rustpython either, because I saw you guys were looking into tracing some performance data (flamechart or how it's called). I should pause here and explicate myself - github does allow to have a good insight into somebody else's activity. And it's not about peeking into somebody's personal stuff. It's more like: that guy started such a wonderful project, and now put it aside and works on some other project. Maybe that's new project is much more cooler? Maybe I should do the same - drop my current projects, which I consider important and work on something else? Or maybe I should stay aside to not feel victim of Stockholm syndrome? ;-).

So, while github does allow to see what others work on, I appreciate you bringing it up in discussion, and a chance to exchange a few words about - heck, who am I to do that, but a random passer-by?? So, well, I'd really appreciate if you could consider value of rustpython vs "some" (and you know which I mean) of your projects. Because what happens when rustpython is "finished"? There will be just yet another Python implementation in a line-up of CPython, Jython, IronPython, PyPy, etc. Don't get me wrong - RustPython is absolutely great project. I write all this only because I'm painfully aware that one never "does something", only "does something instead of something else". And as my clear aim is to motivate you to come back to PPCI, which is a truly unique project, I can't help by to call for consideration of the value of RustPython.

pfalcon, 2019-12-31:
> Part of the reason why I stopped working on this project was my doubt about rust. It's a great language and I was doubting if I should rewrite all of ppci in rust. The performance would be better, [...]

Thanks for explaining this situation. And all I can say - "we all were on the brink abyss" ;-). It's a moment similar to Damien George, the author of MicroPython, to once have confessed  he had considered to write a small implementation of Ruby instead ;-).

So yeah, I really try to stay neutral and considerate while writing this. So, Rust is a great language. But "Rewrite everything in Rust" is a frenzy. There're just too many great languages around. And each has its benefits and drawbacks. The biggest drawback of Python is that it's too usable and too beginner-friendly (or beinner-luring). I painfully learnt that while trying to do user support for MicroPython. But those very traits are the biggest benefits of Python - it's the language which is common ground for both pro's and beginners, and suitable for multitude of tasks.

So, thanks for resisting desire to rewrite PPCI in Rust. Again, it's main selling point is that it's in Python, and thus easily hackable. To give an example of what I think may be target usage of PPCI is: So, in MicroPython, some guy is trying to do low-level optimizations of codegeneration, where desired behavior is dependent on an actual call site to a function (not a function itself): https://github.com/micropython/micropython/pull/5314#issuecomment-556122763 (hello @Jongy). Obviously, with contemporary compilers, any work in that vein (actually changing the C source) would be a useless premature optimization, where a change helps one architecture, but harms another, so there will be just flip-flopping on the source code level back and forth. 

But with project like PPCI, one could do it in right way, by hacking together an adhoc optimization pass, either automatic, or driven by some annotations. Well, that could be done in LLVM. I'd assess time frame as half a year for some tangible results. For a compiler implemented in Python, it should be days (a day for initial hack-up, which will be dirty, but still much more reusable than a corresponding hack-up done to C++ of LLVM).

pfalcon, 2019-12-31:
> Do you have any specific directions you would like to see the project move?

Well, let me humbly remind that I submitted ~20 tickets, covering exactly the things I'd consider important to do (so many, to keep them focused, actionable, then closable). So, if you're interested in contribution, it might be a good idea to go thru and comment on them, to see if we have some areas of common vision (and to let other folks know, that's again why I went thru the trouble of posting a bunch of detailed tickets).

If you would ask me for prioritization, I'd say that https://github.com/windelbouwman/ppci-mirror/issues/24 should be an immediate technical priority. Quoting it:

> I'd humbly suggest that a race to get following scenario work (ppci-cc hello.c -o hello; ./hello) should be the first task for the project.

On the organizational side (not less important), I'd say discussing https://github.com/windelbouwman/ppci-mirror/issues/5 is important. And https://github.com/windelbouwman/ppci-mirror/issues/16 is something to address ASAP. (For me personally, handling of typos is a litmus test of a project activity and approach - it's always my first suggestion "if you want to contribute, fixing a typo is the smoothest way to start", and vice-versa, acting on typo reports is something I pledge to myself to do as regardless as possible of extra matters, like reporter's following some recommended submission process, etc. That's why seeing zero response on that is somewhat disheartening ;-) ).

pfalcon, 2019-12-31:
> So, if you're interested in contributions

Oh, and if you don't, might be helpful no know that and why. No matter what it is (and if it's personal ;-)), from an author of a project like this, I'd personally take it as a learning experience any response.

Thanks again!

windelbouwman, 2019-12-31:
Agreed that one of the strengths of this project is / should be hackability, this was one my initial aims. @Jongy what were you trying to achieve? Did you succeed?

Jongy, 2019-12-31:
First of all, that's a really cool project you got here! I also remember you from RustPython :)

My idea in https://github.com/micropython/micropython/pull/5314 was to "uninline" a single memory access whose assembly representation costs more than a function call, therefore it can be replaced with a function call, which needn't save the registers of the current frame because it's almost always used as the last basic operation in functions - so I hoped TCO will come in handy. It wasn't so effective as I expected for Thumb, because the instruction sequence required for TCO costs more bytes (`POP` into LR and not PC is 4 bytes instead of 2).

It was still somewhat effective, though, as you can see in the PR description. But it was not maintainable and as @pfalcon said, it'll just create flip-flopping between micro changes.

Later in https://github.com/micropython/micropython/pull/5320 the idea was to exploit shorter instructions for data loads of commonly referenced items. It wasn't so maintainable either - this time since it required core changes in the ldscript of each board wanting to use it. Eventually Damien took  another idea I suggested further and created https://github.com/micropython/micropython/pull/5429, which is a change in the data representation providing the same effect, and it's maintainable and portable.

Now - I definitely think a compiler plugin could automatically "think of" the changes in my 2 first PRs, it could test for their effectiveness, and it could then apply them itself. It's just a simple set of rules. 

From my experience writing GCC plugins, I think it'd be too complex for anyone who's not an experienced GCC developer. I haven't taken a deeper look on this project yet, but I believe it might allow just that - any developer with an automation idea can just add passes easily -  which is pretty awesome.

windelbouwman, 2019-12-31:
Allright, interesting usecase! I tried to compile micropython with ppci, but I did not yet succeed with that :). Thanks for the detailed explanations!

pfalcon, 2020-01-01:
Happy New Year everyone on the thread!

I would like to extend an invitation to @CarterTS who just [posted on Reddit](https://old.reddit.com/r/Python/comments/eieuld/c_compiler_written_in_python/) about https://github.com/CarterTS/CCompiler , to consider joining PPCI fun ;-).

windelbouwman, 2020-01-01:
Happy 2020!

windelbouwman, 2020-01-16:
@pfalcon should we close this issue or leave it open?

pfalcon, 2020-01-16:
Well, I don't know, imho this ticket has useful info, and some points to discuss (I'm for one pretty slow, sorry). But decide for yourself, if you think that keeping too many "meta" tickets open doesn't make sense, well...