windelbouwman/ppci #33: List of C compilers written in Python

pfalcon, 2020-01-02:
Some of these were mentioned throughout comments, and I guess it makes sense to have a dedicated, visible list. My motivation behind this list is to avoid situation when some party involved is not aware that other similar projects exist, so any work can be a potential duplication of effort. To that end, I @-mention users involved. My apologies if this information is of no interest to you - there's an "Unsubscribed" button on the right in Github UI to stop receiving further notifications.

* https://github.com/campagnola/pycca by @campagnola
* https://github.com/ShivamSarodia/ShivyC by @ShivamSarodia
* https://github.com/CarterTS/CCompiler by @CarterTS
* https://github.com/windelbouwman/ppci-mirror by @windelbouwman
* https://github.com/shiyanhui/Compiler by @shiyanhui

windelbouwman, 2020-01-10:
I made a section in the documentation listing similar projects, should we move this list over there? Maybe we can also group the projects by theme?

pfalcon, 2020-01-10:
Well, I don't know. I really posted this ticket with the motivations described. I'm not sure if it's useful to have such a list in the docs. From my PoV, addressing https://github.com/windelbouwman/ppci-mirror/issues/21 in docs or README (or both) would be more beneficial.

pfalcon, 2020-01-12:
Maybe there will be port of recent hit of compiler interwebs, "A Compiler Writing Journey", to Python, by @ponyatov: https://github.com/ponyatov/pyccwj