windelbouwman/ppci #35: Improve/make consistent "python3 -m ppci" behavior

pfalcon, 2020-01-03:
I personally would prefer to not rely on entrypoints created by the installer script (and ideally/eventually, not rely on the need for ppci to be "installed" at all).

So, the first thing I'm greeted is:

~~~
$ python3 -m ppci
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/__main__.py", line 18, in <module>
    main()
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/__main__.py", line 10, in main
    subcommand = sys.argv[1]
IndexError: list index out of range
~~~

Now, there's ppci-ld, but:
~~~
$ python3 -m ppci ld
...
ModuleNotFoundError: No module named 'ppci.cli.ld'
~~~

Suddenly, the linker is under `python3 -m ppci link`. Let's fix it up and make it consistent, which is of course "ld", per https://github.com/windelbouwman/ppci-mirror/issues/23 . If there's strong desire "link" can stay as an alias, but I'd recommend against it, let's just just clean up such small cases, it's just not worse to maintain "diversification" in such small things, it will only lead to confusion and maintenance overheads.

windelbouwman, 2020-01-03:
I addressed this issue in 2d2fb521531bad210ff9ef6a1de4ee1909a7547a. Do you know a way to hook existing argparse parsers under an umbrella parser? The way it is done now feels pretty manual.

pfalcon, 2020-01-04:
> I addressed this issue in 2d2fb52.

Thanks for prompt looking into this. But yeah, I'd say the implementation is somewhat sub-ideal: we have a set of modules there, and now we have the ` valid_programs` list which duplicates set of modules. My idea how to implement that was along the lines of calling `os.listdir()` on package's `__path__`. That has its own drawbacks too (won't work with ppci in .zip), but we can't fix all issues at once.

> Do you know a way to hook existing argparse parsers under an umbrella parser?

Well, if I understood you correctly, then yes, it's possible to have "hierarchical" argparser's. I myself never had a need to use that pattern so far, but I remember that esptool uses it: https://github.com/espressif/esptool/blob/master/esptool.py#L2820 .

But, I personally don't immediately grok how it would be helpful here. So, just in case, let me warn against complicating the current state of affairs. Current design of loosely coupled cli modules, plugin-style, with dynamic loading, seems to be the simplest and easiest one. Actually, I'd call that even that dup-list above is "premature optimization". If Python is a dynamic language which allows us to list files in package's directory, why not use that as the low-hanging solution, until reasons for more complex/static solution come? (It's more complex because it requires more code and more maintenance.)

(Oh and btw, just right now I'm working on defining "Python strict mode" ;-). The main motivation is to get rid of symbolic lookups, and thus constraint is not being able to add new items to namespaces at runtime. By implication that means no imports at runtime. So, if you want to disregard my advice above, go ahead, I need application specimens to test my strict mode against ;-). The point remains the same - at the current stage, for PPCI that would be *premature optimization*).

pfalcon, 2020-01-04:
> Suddenly, the linker is under python3 -m ppci link. Let's fix it up and make it consistent, which is of course "ld", per #23 . If there's strong desire "link" can stay as an alias, but I'd recommend against it

Oh, and can you do something about this please? I don't submit this rename commit because I'm not sure about your stance on that, the whole point of discussion in https://github.com/windelbouwman/ppci-mirror/issues/11 is to clarify that matter.

windelbouwman, 2020-01-04:
I added an `ld` alias to `link` to make it consistent with the `ppci-ld` command.

pfalcon, 2020-01-04:
Works for me, thanks!

closed 2020-01-04 by pfalcon