windelbouwman/ppci #36: No way to specify program entry point, ELF hardcodes at 0x40000

pfalcon, 2020-01-03:
D'oh: https://github.com/windelbouwman/ppci-mirror/blob/master/ppci/format/elf/file.py#L219

This in turn leads to crazy workarounds like:

~~~
// No idea how to specify entrypoint to ppci linker, so far, entry point is
// at the beginning of the segment, and functions appear in the segment in
// the order of appearance in the source file, even if "apperance" is merely
// a prototype.
~~~

If not for inline asm, added just yesterday, it wouldn't be possible to workaround that at all (in "single source file" model, I mean).

pfalcon, 2020-01-03:
So, please let's make sure that this gets implemented properly. This option is the option to `ld`:

~~~
       -e entry
       --entry=entry
           Use entry as the explicit symbol for beginning execution of your
           program, rather than the default entry point.  If there is no
           symbol named entry, the linker will try to parse entry as a number,
           and use that as the entry address (the number will be interpreted
           in base 10; you may use a leading 0x for base 16, or a leading 0
           for base 8).
~~~

Then apparently `ld` will need to store it in .oj file format. Likely, in resolved form (sumbol -> address). And then tools like `objcopy --output-format elf` will use it to fill in actual headers.

windelbouwman, 2020-01-09:
Initial steps are done here: c078ba4310dec64600ddb78a35e79fead16cd43e

TODO: add the --entry flag to ld.

windelbouwman, 2020-01-09:
Command line argument implemented here: 0043061b09872c1ac2ab98d7b9e9ba4b10473f80

pfalcon, 2020-01-10:
> Command line argument implemented here: 0043061

Thanks! Confirming it works as expected. Closing.

closed 2020-01-10 by pfalcon