windelbouwman/ppci #43: IR questions/criticism

pfalcon, 2020-01-13:
My attempt to start looking inside the project, not just at the surface of it.

~~~
    blob<8:8> alloc_a = alloc 8 bytes aligned at 8;
    ptr addr_a = &alloc_a;
    store i_phi, addr_a;
~~~

Why "alloc" operation returns "an object" instead of pointer to object? What is semantics of this operation, how to reason about it? Read literally, it's return by value - "allocate object somewhere, then return its value". That doesn't make any sense. At best, this requires 3 lines to do a common operation like above, instead of 2 (taking an address is superflous).

That "allocate somewhere" is also confusing. If it allocates on the stack, why not call it classically, "alloca". Then there could be separate "heapalloc" instruction, if it ever comes to that (you'd definitely need that to high-level'ly compile a language with automatic memory management, like Python).

pfalcon, 2020-01-13:
~~~
    i64 one = 1;
    i64 i_inc = i_phi + one;
~~~

Why any IR output I see again assigns any constant to temporary first? That again makes twice more lines than needed, and makes IR almost human unreadable. Is that limitation of IR or just of how it's currently generated? In either case, I suggest to fix it sooner rather than later.

pfalcon, 2020-01-13:
ir.Return / ir.Exit split is considered harmful. Even the main developer is confusing it:

~~~
def foo():
    return 1
~~~

~~~
python3 -m ppci pycompile --ir ret.py
...
    assert isinstance(function, ir.Function), type(function)
AssertionError: <class 'ppci.ir.Procedure'>
~~~

So, pycompile instantiates ir.Procedure for a python function, but then stuffs ir.Return into it. All this artificial type mismatching is rather unneeded.

pfalcon, 2020-01-13:
~~~
        if return_type:
            ir_function = ir.ExternalFunction(
                name, ir_arg_types, self.get_ty(return_type)
            )
        else:
            ir_function = ir.ExternalProcedure(name, ir_arg_types)
~~~

This is totally unneeded.

pfalcon, 2020-01-13:
> Even the main developer is confusing it

Ah, so it's me who's confused after all, I didn't provide return type annotation, and ppci decided that it's "procedure", even despite the fact that in Python all functions return value. So, well, not sure I will be able to deconfuse that ;-).

pfalcon, 2020-01-13:
Question: how to get IR after SSA conversion but before any other optimizations? Logically, ppci-opt would do that, but I don't ability run specific passes.

OTOH, running naive IR thru ppci-opt -O2 leads to:

~~~
global function i64 foo(i64 a) {
  foo_block0: {
    i64 num_0 = 0;
    cjmp a != num_0 ? foo_block2 : foo_block3;
  }

  foo_block2: {
    i64 num_1 = 1;
    i64 augassign = a + num_1;
    return augassign;
  }

  foo_block3: {
    i64 num_3 = 0;
    return num_3;
  }

}
~~~

I.e., alloc sinking is done, which is kinda advanced optimization. And yet there're multiple returns! Despite the fact that many optimizations require single entry/single exit CFG. Heck, codegeneration requires 1e/1e CFG due to prolog epilogue stuff. So, how to understand that? Single-exit'ing was done, and then even undone? To burden codegeneration backend with doing adhoc version of it again? Or it wasn't done in the first place?

It seems that SSA generation in the presence of return's is somewhat simplified. But I'm eager to find bugs in PPCI's SSA handling, and so far, it's all elusive, it cunningly hides them! :-D

pfalcon, 2020-01-13:
~~~
    i64 phi_alloc_b_1 = phi foo_block0: num, foo_block6: num_6;
    i64 phi_alloc_a_1 = phi foo_block0: a, foo_block6: augassign;
~~~

This is also pretty unreadable. Either given that, or despite that I'm pretty trained to read LLVM IR. When you see stuff like "phi_alloc_b_1", then: a) it's too long; b) your mind tries to assign meaning to all that. But there's hardly any meaning, so all those "phi", "alloc", "b" are just confusing. Or perhaps I'm just not trained to see the meaning of it. But I would bet on the fact that there's no useful meaning in it. It's just a temporary. And a natural way to name temporaries is "tmp0'', "tmp1", "tmp2", etc. Well, maybe even that would be too long. There's a reason why LLVM uses %0, %1, etc.

windelbouwman, 2020-01-14:
> Why any IR output I see again assigns any constant to temporary first? That again makes twice more lines than needed, and makes IR almost human unreadable. Is that limitation of IR or just of how it's currently generated? In either case, I suggest to fix it sooner rather than later.

This is done so that binary operation can deal with any value, there is no special case for the constant value. I agree that this leads to IR code clutter, but is also indicates that there is a const instruction involved, otherwise the const instruction would be hidden in the binary operator, store operator etc..

> r.Return / ir.Exit split is considered harmful. Even the main developer is confusing it:

This was done to prevent the void type. It's an idea from pascal I think in which there is no void type and there are procedures and functions to make the distinction. The error you are facing here is actually a bug in the python to ir compiler.

> Why "alloc" operation returns "an object" instead of pointer to object? What is semantics of this operation, how to reason about it?

This was a change I added later, when functions could return structs by value.

> Question: how to get IR after SSA conversion but before any other optimizations? Logically, ppci-opt would do that, but I don't ability run specific passes.

I feel this would be a nice addition to the `ppci-opt` utility?

> .e., alloc sinking is done, which is kinda advanced optimization. And yet there're multiple returns! Despite the fact that many optimizations require single entry/single exit CFG. Heck, [...]

Yes, this optimization is pretty cool :). The control flow graph (CFG) requires a single entry and a single exit. I think there is a hidden exit node which is targeted by the exit and return instructions.

> It seems that SSA generation in the presence of return's is somewhat simplified. But I'm eager to find bugs in PPCI's SSA handling, and so far, it's all elusive, it cunningly hides them! :-D

Please continue your attempts, I would highly value this!

> This is also pretty unreadable. Either given that, or despite that I'm pretty trained to read LLVM IR. When you see stuff like "phi_alloc_b_1", then: a) it's too long; b) your mind tries to assign [...]

The temporary names are pretty random, they reflect now the instruction which creates the value. Do you think tmp1, tmp2 etc.. would be an improvement?

pfalcon, 2020-01-14:
Thanks for the response!

>> Why any IR output I see again assigns any constant to temporary first?

> This is done so that binary operation can deal with any value, there is no special case for the constant value.

I see, but I argue, that the effect is the opposite: binary operations in your IR **can't** deal with any value, only with subset of values, namely virtual registers.

Because a set of primary values (operands) of instructions is:

* Virtual registers
* Constants

(We can also mention labels (which is essentially a constant either, but whose value is known only at the link time).

Limiting that to vregs isn't rooted in problem domain. Because you see, a typical CPU does have both instructions of type "ADD EAX, EBX" and "ADD EAX, 2".  Trying to prepend that only first type exists has grave implications on IR processing. We already discussed human readaility issue, but there're  technical ones.

For example, how do you deal with simplifying arithmetic identities like "a = b + 0" => "a = b"? With IR which contains constants, that's trivial O(n) algo. With your IR, it's either O(n^2) or complex algo which builds maps which track which vreg is actually a constant.

Then how your constant/copy propagation passes work? Because any such pass would turn:

~~~
    i64 one = 1;
    i64 i_inc = i_phi + one;
~~~

into:

~~~
    i64 one = 1;
    i64 i_inc = i_phi + 1;
~~~

So, do you want to say that you don't have such passes? But those are the simplest and most fundamental ones.

Then finally, instruction selection. Again, there're machine instructions of "add reg, reg" vs "add reg, imm" types, and matching the latter against `i64 i_inc = i_phi + 1` is much easier.

So, the only use from stuff like `i64 one = 1` I see is that it kinda gives a type of constant. But in most cases, it's implicit, e.g. type of addition is known by type of result, and both operands should be of the same type. Sometimes, you can't rely on implicit types, e.g. in arguments to function calls, so there should be syntax to type constants (if you don't have one already).

pfalcon, 2020-01-14:
> The temporary names are pretty random, they reflect now the instruction which creates the value. Do you think tmp1, tmp2 etc.. would be an improvement?

Yep, a short uniform naming for temporaries would be a good improvement. (So, `tmp1`, `t1`, `_1`, `$1`, etc. would be considered). I'm open to counter-arguments actually, maybe I'm biased and miss something. But we have LLVM IR as a reference. And don't get me wrong - LLVM IR is rather not ideal IR, but it's very easy to be worse than it (actually, there're tons of IR like that). The point is to be better ;-).

pfalcon, 2020-01-14:
>>     r.Return / ir.Exit split is considered harmful. Even the main developer is confusing it:

> This was done to prevent the void type. It's an idea from pascal I think in which there is no void type and there are procedures and functions to make the distinction.

But that's again, progress backwards. It's a huge achievement of human mind to arrive at idea that "no type" is just a special kind of type, not something completely unearthly which requires a special notion to be dealt with. Wirth was afraid to teach his students that, but there's no need to repeat his mistakes. It again leads to deep implications, like 2 different types of code containers (functions vs procedures) now require 2 different types of termination statements (return vs exit). And dealing with those dichotomies is significant burden comparing to just dealing with one of many return types (even if void is a little bit special, but only a tiny little bit, in that it offers more opportunities for optimization).

tstreiff, 2020-01-15:
I agree that current IR is hard to read, harder than LLVM anyway. Using more generic identifiers for vregs would help Indeed. I tried to find a meaning in the current ids, but they can be sometimes confusing, I have seen :
i64 constant_0 = 2
I was surprised to see that IR instructions do not natively support constants. This creates many additional vregs and an additional level of indirection. Fortunately this is not reflected in the trees to be matched by the backend :
vreg1 = constant(1)
store vreg1 (stack -2)
leads to a tree like :
STRI64 FPREL(-2) CONSTI64(1)
so some changes can be done in the IR without breaking the backend interface compatibility.
(however I have seen that the x86_64 backend does not use instructions between registers and constants)
Functions returning nothing sounds better than procedures. Just because most of the required code to process functions and procedures is the same.

windelbouwman, 2020-01-15:
@almarklein and @michg, I'm curious to your opinions about this as well. In general modifying the IR can be pretty hairy, since both frontends and backends must be modified accordingly. That's why I would like to have as many opinions as possible, before making a decision.

To make the IR-code easily adaptable, I would like to add more functions to the `irutils.Builder` class, such that this class can emit different instructions and pick names for them, like suggested here.

windelbouwman, 2020-01-15:
Btw, so far I'm in favor of naming all variables `tmp1`, `tmp2` etc. I can see this cleaning up the IR significantly, also the inline constants I agree, this should probably be done.

The procedures/functions vs void tradeoff is somewhat hard for me, since I was happy losing the artificial void type in the first place. Since most code already would have to use code like `if return_type is void`, I figured I might as well change this code into `if routine.is_procedure`.

almarklein, 2020-01-16:
*Readability counts.* That said, my experience with ppci IR is very limited, so I trust your opinion on this one :)

pfalcon, 2020-03-16:
Getting back to this point:

> I was surprised to see that IR instructions do not natively support constants. This creates many additional vregs and an additional level of indirection.

I read https://developers.redhat.com/blog/2020/01/20/mir-a-lightweight-jit-compiler-project/, and it utilizes a similar approach. It's described in the https://developers.redhat.com/blog/2020/01/20/mir-a-lightweight-jit-compiler-project/#mir-generator section:

> First, we simplify MIR as much as possible, which means only using register indirect memory addressing. It also means that immediate instruction operands are used only in move instructions.

The key difference: in MIR, that's a separate pass, as executed in a codegeneration process. In other words, the input IR is in human-readable form like `add i, i, 1` (based on the example in that article), and only code transformation pass turns that into something verbose like:

~~~
one = 1
add i, i, one
~~~

pfalcon, 2020-03-22:
Another reference on the matter: https://flex.cscott.net/Harpoon/quads/node3.html :
 
> In addition, the QuadSSA form is maximally factored. Constants are not allowed as quadruple operands (except for a special const operation); which creates a unique mapping from variable names to values in the computation. This simplifies value analysis. 

It's unclear whether this "QuadSSA" form is an internal transformed representation for some passes or externally-facing representation.

pfalcon, 2020-05-30:
BRIL (https://capra.cs.cornell.edu/bril/langref.html) is another IR which doesn't allows immediates in instructions taking multiple args:

> A Constant is an instruction that produces a literal value.

> Value Operation
> args, a list of strings. The strings are names interpreted according to the operation. They may refer to variables or labels.

pfalcon, 2020-10-08:
Here's another big question. All this time I assumed that PPCI IR doesn't necessary have to be in SSA, i.e. that PPCI builds SSA as an explicit pass, on standard register variables. However, it seems to be nor the case. Feeding following IR to ppci-opt leads to error:
```
module main;

global function i32 add(i32 a, i32 b) {
  add_block0: {
    i32 num_11 = 0;
    cjmp b == num_11 ? add_block4 : add_block3;
  }

  add_block3: {
    i32 num_1 = 2;
    i32 tmp = a + num_1;
    jmp add_block2;
  }

  add_block4: {
    i32 num_3 = 1;
    i32 tmp = a + num_3;
    jmp add_block2;
  }

  add_block2: {
    return tmp;
  }
}
```

Error is:

```
Traceback (most recent call last):
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/.venv/bin/ppci-opt", line 11, in <module>
    load_entry_point('ppci', 'console_scripts', 'ppci-opt')()
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/cli/opt.py", line 20, in opt
    api.optimize(module, level=args.O)
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/api.py", line 233, in optimize
    verify_module(ir_module)
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/irutils/verify.py", line 20, in verify_module
    Verifier().verify(module)
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/irutils/verify.py", line 36, in verify
    self.verify_function(function)
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/irutils/verify.py", line 99, in verify_function
    self.verify_block(block)
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/irutils/verify.py", line 117, in verify_block
    self.verify_instruction(instruction, block)
  File "/mnt/hdd/projects-3rdparty/Python/Python-compilers-for-non-python/ppci-mirror/ppci/irutils/verify.py", line 180, in verify_instruction
    ), "{} does not dominate {}".format(value, instruction)
AssertionError: i32 tmp_0 = a + num_3 does not dominate return tmp_0
```

So, it seems that PPCI follows the same path as LLVM, where IR is forced to be in SSA, and a workaround for that is to go roundabout way of explicitly allocating variables on stack and accessing them with loads/stores. So, 2 extremes - one complex, another bloated and no middle-ground between them, where variables are just variables, but can be multiply assigned to (i.e., how most programs out there are written).

Can you please confirm this @windelbouwman or let me know if I'm missing something. Thanks!