windelbouwman/ppci #44: 8bit addition générated

tstreiff, 2020-01-13:
Hello,
When compiling the following simple function with ppci-cc 0.5.8
int f(char *s) {
  return *s - '0';
}
The generated IR code is
global function i64 f(ptr s) {
  f_block0: {
    i8 load_0 = load s;
    i8 constant = 48;
    i8 op = load_0 - constant;
    i64 typecast = cast op;
    return typecast;
  }
}
The C standard specifies that operands narrower than ints are extended to int width before an operation can be done. However optimisations are possible as far as the result is consistent with this rule.
In our case, doing an 8bit subtraction can trigger an integer underflow and lead to an incorrect result. 
Example if *str is -120 (0x88), doing an 8bit subtraction with 48 (0x30) will make 0x58 which is +88. Then with the extension to i64 the function will return +88.
If the 2 operands are first extended to ints before subtraction, the result becomes correct : -168.

pfalcon, 2020-01-13:
+1, I believe I saw something similar too.

windelbouwman, 2020-01-14:
Thanks! I was unaware of this. The C frontend must be extended to handle this, and a test snippet demonstrating this example must be added.