windelbouwman/ppci #48: ppci-cc : using [] in a function argument makes the compiler crash

tstreiff, 2020-01-15:
When compiling the C code below :

> void print_board(register int board[][8]) {
> }

ppci-cc raises a Python exception AttributeError: 'ArrayType' object has no attribute 'location'
Here is the progress report :

> 2020-01-15 23:14:21,486 |     INFO |       root | ppci 0.5.7 on CPython 3.6.9 
> 2020-01-15 23:14:21,492 |     INFO |   cbuilder | Starting C compilation (c99)
> 2020-01-15 23:14:21,493 |     INFO |    cparser | Parsing finished
> AttributeError

This happens during IR code generation when the compiler tries to evaluate the size of the "board" argument. It fails because 1st size is not defined (which is legal for a function argument). It tries to report the error but type.location is apparently not defined.

2 problems here :
- [] is apparently not supported in function arguments (C standard states that "array of type" in arguments are considered as "pointer to type" and therefore int board[][8] is the same as int (*board)[8]
- the error reporting raises an exception

windelbouwman, 2020-06-07:
This is addressed in ee5939d37ceab7e157a706e92632480ba01df3c5

@tstreiff could you recheck the behavior with your libc?

tstreiff, 2020-06-22:
It now works, the issue may be closed.

closed 2020-06-22 by tstreiff