windelbouwman/ppci #49: ppci-cc : problems handling static string initializers

tstreiff, 2020-01-15:
Compilation of the following C source code went well but the generated bytes are wrong (comments mention expected results).
> char c0[] = "012";      // size is 4
> char c1[9] = "012";     // 3 digits + 6 trailing zero bytes
> char c2[3] = "012";     // 3 digits and no trailing zero bytes
> char c3[4] = "012";     // 3 digits + 1 trailing zero byte
> char c4[3] = "012345";  // should report an error (initializer too long)
> char c5[3] = { '0', '1', '2', '4', '5' };  // should report an error (initializer too long)

1) General serious issue : the " string delimiters (ASCII code 34) are included in the generated bytes.
Eg: c0 generates bytes 34, 48, 49, 50, 34, 0 instead of 48, 49, 50, 0

2) Generated number of bytes for c2, c3, c4, c5 is wrong : when a size is specified for an array, this is the actual size of the variable, and the intializer cannot change that.

3) There are warnings on the size in the progress report but rather well hidden:

> 2020-01-16 00:22:18,296 |     INFO |       root | ppci 0.5.7 on CPython 3.6.9
> 2020-01-16 00:22:18,303 |     INFO |   cbuilder | Starting C compilation (c99)
> 2020-01-16 00:22:18,310 |  WARNING |   ccontext | Excess elements!
> 2020-01-16 00:22:18,310 |     INFO |   ccontext | At: (bug8.c, 8, 31), hints: None
> 2020-01-16 00:22:18,311 |  WARNING |   ccontext | Excess elements!
> 2020-01-16 00:22:18,311 |     INFO |   ccontext | At: (bug8.c, 8, 36), hints: None
> 2020-01-16 00:22:18,312 |  WARNING |   ccontext | Excess elements!
> 2020-01-16 00:22:18,313 |     INFO |   ccontext | At: (bug8.c, 8, 41), hints: None
> 2020-01-16 00:22:18,314 |     INFO |    cparser | Parsing finished
> 2020-01-16 00:22:18,315 |     INFO |   ccodegen | Finished IR-code generation

windelbouwman, 2020-01-16:
I'll give this issue a try!

windelbouwman, 2020-01-16:
> char c4[3] = "012345"; // should report an error (initializer too long)
> char c5[3] = { '0', '1', '2', '4', '5' }; // should report an error (initializer too long)

Should this be an error, or a warning?

tstreiff, 2020-01-16:
For me the ideal behaviour would be to truncate the initializer to the variable size and issue a warning (because the resulting code will be correct)
(most compilers including gcc do that)

windelbouwman, 2020-01-16:
Okay, this is how I implemented in def31436e961e86cc5bae251bc7d7d0d495aadd3
The warning is not properly displayed yet. I think we need a better warning system.