windelbouwman/ppci #52: ppci-ci: preprocessor does not accept empty macro argument values

tstreiff, 2020-01-22:
> #define M(arg) arg
> #define CALL(f, ...)    (f)(__VA_ARGS__)
> void f() {
>   M(;)                        // accepted
>   M()                         // rejected
>   CALL(g0);                // rejected
>   CALL(g1, 1);            // accepted
>   CALL(g2, 1, 2);        // accepted
>   CALL(g3, 1, 2, 3);    // accepted
> }
However it is legal to invoke a macro with empty arguments (ie with no tokens) as long as the number of commas is correct for the macro.

2020-01-22 21:44:46,002 |     INFO |       root | ppci 0.5.7 on CPython 3.6.9 on Linux
2020-01-22 21:44:46,010 |    ERROR |       root | Got 0 arguments ([]), but expected 1
2020-01-22 21:44:46,010 |    ERROR |       root | None
Got 0 arguments ([]), but expected 1
The current error message does not mention that the problem is in a macro invocation (the source location of the error and the name of the macro would be useful)

windelbouwman, 2020-02-16:
This should now work out, as per this commit: 08d77f031b279d587b8364e957325228dfa3e1db

tstreiff, 2020-06-08:
Just checked: OK!

closed 2020-06-09 by windelbouwman