windelbouwman/ppci #54: ppci-cc: in x86_64, "long" types are smaller than "int" types

tstreiff, 2020-01-30:
I am trying to compile my own small libc with target x86_64.
I have just realized something weird: sizeof(int) > sizeof(long)
ints are 8-byte long in x64, and I expected longs to be the same, but they are only 4-byte long.
The C standard requires that sizeof(int) <= sizeof(long) but this is not the case.
According to the code I have read, it seems that "longs" are 32bit on all targets. "ints" size is defined in the arch.py file of the target.
Fixing this requires that the size of longs is taken as max(32, sizeof(int))
I tested this by changing context.py as follows and it works (longs are at least 32bit as required by the C standard but cannot be smaller that ints)
> self.type_size_map = {
>             BasicType.CHAR: (1, 1),
>             BasicType.UCHAR: (1, 1),
>             BasicType.SHORT: (2, 2),
>             BasicType.USHORT: (2, 2),
>             BasicType.INT: (int_size, int_alignment),
>             BasicType.UINT: (int_size, int_alignment),
**>             BasicType.LONG:  (max(int_size, 4),  max(int_alignment, 4)),
>             BasicType.ULONG: (max(int_size, 4),  max(int_alignment, 4)),**
>             BasicType.LONGLONG: (8, 8),
>             BasicType.ULONGLONG: (8, 8),
>             BasicType.FLOAT: (4, 4),
>             BasicType.DOUBLE: (double_size, double_alignment),
>             BasicType.LONGDOUBLE: (10, 10),
>         }
As a protection, same change should be done for "long long" for the future 128bit ints CPU...

windelbouwman, 2020-02-12:
I implemented this as per your suggestion.

windelbouwman, 2020-06-07:
Closing this issue, please re-open if required.

closed 2020-06-07 by windelbouwman