windelbouwman/ppci #56: ppci-cc cpp stringify operator # expands too simplistic

tstreiff, 2020-02-11:
The preprocessor stringify operator # is supported but the resulting expansions are not always correct.
(behaviour for 0.5.7)
> #define S(a) #a
> S(word) => "word"            // correct
> S("string") => ""string""    // instead of "\"string\"" (double quotes should be escaped)
> S('a') => "'a'"                      // correct
> S("aa\tbb") => ""aa\tbb"    // instead of  "\"aa\\tbb\"" (backslashes should be escaped)
> S(  1) =>> "  1" au lieu de "1"    // leading and trailing blanks should be ignored
> S(  1  2      3  ) => "  1  2  3" au lieu de "1 2 3"    // only one separator is kept between tokens

windelbouwman, 2020-02-12:
Wow, nice catch. The preprocessor never ceases to amaze me :).

Implemented a fix in 26239aad6fec269b2aa34724dece80351da63761

tstreiff, 2020-02-13:
The fix looks good.
# is tricky but ## is even worse!

windelbouwman, 2020-06-05:
Closing this issue, since it seems to operate properly.

closed 2020-06-05 by windelbouwman