windelbouwman/ppci #58: ppci-cc: cpp #if expression mis-calculated in some cases

tstreiff, 2020-03-16:
In some cases, the result of #if expressions is wrong. I have noticed 2 cases :

1) && and || do not return 0 or 1
The following expressions are always false:

> #if 7 && 3 == 1   (7 && 3 is evaluated as 3)
> #if 7 || 3 == 1       (7 || 3 is evaluated as 7)
> #if 0 || 3 == 1       (0 || 3 is evaluated as 3)

2) Binary operators with same priority are evaluated from right to left

> #if 2*3/2 !=3     is false because 2*3/2 is evaluated as 2*(3/2) yielding 2

I have fixed the 2 problems with the 2 following (small) changes:

1) in lang/c /preprocessor.py in _eval_tree(), at the end of evaluation of both || and &&, add the following lines:

>  if value:
>      value = 1

2) in lang/c/preprocessor.py in parse_expression(), when parsing a new operator, the following code tests if it has to be evaluated after of before the previous one :

>  if left_associative and (op_prio >= priority):
>                     pass

>= must be >

windelbouwman, 2020-06-05:
This should be fixed as per commit 5338bb03886483603d37d6331d3d0a7004ef312a .

Note that you require parenthesis around this example:
`#if (7 && 3) == 1`
since `&&` has lower priority than `==`.

windelbouwman, 2020-06-06:
Closing this issue, please re-open if things do not work as expected.

closed 2020-06-06 by windelbouwman