windelbouwman/ppci #64: New release?

pfalcon, 2020-05-25:
It looks like enough functionality landed in the PPCI to warrant a new release. (Which is always a chance to announce it and spread a word about the project). 

And IMHO, relocatable ELF is pretty big feature to warrant 0.6 version number.

pfalcon, 2020-05-25:
If this sounds good, I'd suggest to finally decide whether project on github here is "first class". It seems it is, and then it makes sense to rename it from "ppci-mirror" to just "ppci". This is minor, but still a point of confusion for users. And no worries, in the case of renaming, github will automatically redirect all the older URLs (up to and including git clones), so there will be zero nuisance. Thanks.

pfalcon, 2020-05-25:
And minor doc view suggestion: on https://ppci.readthedocs.io/en/latest/, at the top left corner, shows Github "Watch" counter. It would be nice to replace it with "Stars" counter (after all, we should motivate users to star the project, only few will be interested to watch it closely).

I looked at preparing a patch for that myself, but apparently, not all Sphinx templates are in the repo, https://github.com/windelbouwman/ppci-mirror/tree/master/docs/_templates has only 1 file, while conf.py lists 4: https://github.com/windelbouwman/ppci-mirror/blob/master/docs/conf.py#L164

windelbouwman, 2020-05-26:
A new release sounds good! It has been a while. I wanted to test some things before releasing:
- riscv samples run via qemu instead of verilator emulated cpi
- web assembly spec test contains some unsolved issues

Given the sheer size of the project, this will never be finished, but those two things are high on my wish list.

windelbouwman, 2020-05-26:
> If this sounds good, I'd suggest to finally decide whether project on github here is "first class". It seems it is, and then it makes sense to rename it from "ppci-mirror" to just "ppci"

Agreed, will update this shortly.

pfalcon, 2020-05-31:
@windelbouwman: Thanks for making suggested updates of renaming the project slug and showing stars counter in the docs!

> * riscv samples run via qemu instead of verilator emulated cpi
> * web assembly spec test contains some unsolved issues

I appreciate such attention to detail, and wish I could join in with such endeavors. But testing is a bit boring matter, and there's usually a long list of what *could* be tested. I think that PPCI has a really good regression testsuite, so as long as it passes on master, it's not a case of release blocker.

In other words, I suggest to remember old good "release early, release often" motto. Anything which didn't fit into this release, can be done in the next release, hopefully not too distant ;-).

pfalcon, 2020-06-08:
@windelbouwman: Any progress on the *release* matter? (Not on fixes, which will be around forever, just on bumping version numbers and making a git tag. ;-).

windelbouwman, 2020-06-08:
Great progress! See here: https://pypi.org/project/ppci/#description

Version 0.6 will come, I would like to safe digits before reaching 1.0 (which should be able to compile CPython).

closed 2020-06-09 by windelbouwman

pfalcon, 2020-06-11:
>  A more interesting question is posting on Hacker News

Ok, posted, please upvote: https://news.ycombinator.com/item?id=23487023