windelbouwman/ppci #66: ppci-cc: missing pointer arithmetic for operators -, +=, -=

tstreiff, 2020-06-07:
Pointer arithmetic for + is implemented (pointer + integral or integral + pointer) but other cases are missing.

This should be as follows:
pointer to T - integral: as for +, ie the pointer address is incremented by "integral" times sizeof(T)
pointer1 to T - pointer2 to T: this yields an integral expression equal to (address1-address2)/sizeof(T)
(both pointers shall point to the same type)

+= and -+ should have similar behaviours, since exp1 .= exp2 "means" exp1 = exp1 . exp2.
pointer1 to T += integral   => adress1 += integral * sizeof(T)
pointer1 to T -= integral   => adress1 -= integral * sizeof(T)

Care should be taken if the pointer types are (void*), because sizeof(void) is not defined. This should be flagged as an error.

pfalcon, 2020-06-07:
> Care should be taken if the pointer types are (void*), because sizeof(void) is not defined. This should be flagged as an error.

Can think about GCC extensions compatibility right away and flag as warning, but treat as char*.

tstreiff, 2020-06-08:
I'll make a pull request for this tomorrow or the next day.

tstreiff, 2020-06-12:
I attach a test program that test all combinations (+ void* "promotion" to char* when applicable)

I noticed something peculiar in a specific case:

> void *pv;
> int *pi;
> diff = pv - pi;
> 
>   149:  diff = pv - pi;
>                   ^ Operands must point to compatible types, got char *  and int *
> 
> 
Reporting an error is correct in this case but the message reports that "pv" has type "char *" which is not right.
In semantics/on_binop, we first ensure both types are not void* (and cast them implicitly to char*) before testing they both points to the same type.

I think that doing the reverse is better: first, ensure both types are the same, and if both are void*, do as if they were both char*

I have checked and GCC does it this way:
t27.c:149:13: error: invalid operands to binary - (have ‘void *’ and ‘int *’)
   diff = pv - pi;
             ^
but if both variable are void*, it stays silent and merely subtract the addresses (as for char*)
[ptr_arith_test.txt](https://github.com/windelbouwman/ppci/files/4771738/ptr_arith_test.txt)