windelbouwman/ppci #67: ppci-cc: test expressions should not be implicitly casted to int

tstreiff, 2020-06-08:
Test expressions also called "conditions" are casted to "int" in the C front-end.
This leads to wrong IR code when testing types larger than ints.
On an architecture where ints are 32bit, and pointers and longs are 64bit, pointers and longs are tested only on their 32bit lower part.
The problem includes expressions in if, while, do/while, and operand expressions of ! && ||.
Examples:
FILE *f;
if (f) ...
if (f && f->fd...)
On both tests, f is first casted to "int" and then tested against 0.

A work-around is to explicitly test against NULL/0:
Examples:
FILE *f;
if (f != NULL) ...
if (f != NULL && f->fd)

In semantics.py, removing the int-coercion in all these cases solves most of the problem.

windelbouwman, 2020-06-08:
This issue is addressed in 1d115ce24d396d1baf4a9c1b7e41ade700242a38 @tstreiff could you verify this is what you expect?

tstreiff, 2020-06-08:
I have just tested it with mixed results:
- it is OK for longs, they are no longer casted to int, and a full 64bit comparison is made
- it is KO for pointer, they are still casted to i32 before the comparison, which is wrong

I quickly checked your changes, this comes from "check_condtion()":

> if not condition.typ.is_integer:
>             condition = self.coerce(condition, self.get_type(["int"]))

You should coerce to int only chars and shorts (that's integer promotion and it may be done elsewhere), other types should be kept.

Later on, in gen_condition(), if the condition is not a boolean operator (==, <, !...), you force a comparison to 0 and this will do the job.
A caution in gen_condition for the ! operator: it indeed inverts the yes and no_blocks, but...
Expr and !exp can be of different types (eg; if expr is ptr, !ptr is an int), so if we remove the !, we have to return to the type of expr and not keeping int.