windelbouwman/ppci #68: ppci-cc: wrong handling of storage-class when a symbol redeclared

tstreiff, 2020-06-08:
Problem 1 (minor):

> static int i;
> ...
> int i;
> 
=> no error is reported, whereas this is a change of linkage (a storage-class transition from "static" to None is forbidden) (C standard section 6.2.2)

Problem 2 (major): classical way of declaring a variable in many include files, and to define it in only one .c file

> extern int errno;	/* typically in include files */
> ...
> int errno;              /* in only one source file */

=> no error is reported, but the definition of "errno" is ignored and not put in the object file.
This means there is no way to define in a file 1 a variable which is declared in an file 2 included by file 1. (C standard section 6.2.2)

I need to reconfigure my environment to be able to make pull requests. The fix being local to only one function, I include it below:

lang/c/semantics.py, function check_redeclaration_storage_class

>     def check_redeclaration_storage_class(self, sym, declaration):
>         """ Test if we specified an invalid combo of storage class. """
>         old_storage_class = sym.declaration.storage_class
>         new_storage_class = declaration.storage_class
>         # None == automatic storage class.
>         # TSF 49: add (static, None) because linkage cannot change from internal to external
>         invalid_combos = [(None, "static"), ("extern", "static"), ("static", None)]
>         combo = (old_storage_class, new_storage_class)
>         if combo in invalid_combos:
>             message = "Invalid redefine of storage class. Was {}, but now {}".format(
>                 old_storage_class, new_storage_class
>             )
>             self.invalid_redeclaration(sym, declaration, message)
> 
>         # TSF 49: if new storage-class is "extern", keep existing storage-class
>         # if not declaration.storage_class:
>         #    if sym.declaration.storage_class:
>         #        declaration.storage_class = sym.declaration.storage_class
>         if new_storage_class == "extern":
>             declaration.storage_class = old_storage_class

pfalcon, 2020-06-08:
> I need to reconfigure my environment to be able to make pull requests.

What kind of recocnfiguration to you have in mind? Just in case, as long as you forked project on github, you're about set to submit pull requests. The only remaining ingredient is to make sure you make your changes on separate branches, not on the "master" branch ("master" belong to upstream). Then you have a branch per PR (such branches are called topic branches).

Once that work, the only thing to make it all perfect is to add "upstream" git remote pointing the main repository, and rebase on upstream to pull in new changes into your fork.

tstreiff, 2020-06-08:
Not much, but what I did initially is just copy the ppci tree.
My initial intention was simply to start from this and develop a back end.
I was not sure of the status of the PPCI project.

As I faced compiler issues, I started to patch my local directory to make progress.
I have my own C compiler translated from C to Python so most issues I have found look familiar.

So what I need to do is to save my local directory (containing my changes), reclone locally the current latest PPCI version and recheck my fixes on separate branches to be able to submit pull requests.

Relaigning with "master" branch will also allow me to test the fixed issues.

Thanks for your advice!