windelbouwman/ppci #77: ppci-cc: wrong IR code for function indirect calls

tstreiff, 2020-06-09:
> void (*pf)(int);     // pointer to a function having one argument of type int
> 
> void f() {
>   (*pf)(2);
> }

generates:

>     i32 num_0 = 2;
>     ptr tmp_load = load pf;				get pf contents which is the function address
>     ptr tmp_load_1 = load tmp_load;                  ???
>     call tmp_load_1(num_0);                                indirect call

Even clearer in x86 assembly language:

>        mov rax, pf                           rax is address of pf
>        mov rax, [rax]                        rax is contents of pf (address of the function)
>        mov rax, [rax]                        ??? extraneous indirection
>        mov edi, 2
>        call *rax                                 crash

windelbouwman, 2020-06-13:
This is fixed  in this commit d8c4d2b824dabb5a737c5f530454f38d8c3bda07

Not fully sure about the solution though. I added also some test cases.

tstreiff, 2020-06-13:
Good! My tests now works.
I now be able to test bsearch and qsort in my clib.

closed 2020-06-14 by windelbouwman