windelbouwman/ppci #78: ppci-cc: in "for" loops, a "continue" bypasses the increment expression

tstreiff, 2020-06-09:
The following program loops forever:

> int i;
> for (i = 0; i < 10; ++i)
> .. if (i ==0) continue;

The root cause is that in the current IR generation, a "continue" jumps to the loop condition without executing the iterator/increment expression "++i".
The bug is located in codegenerator.py/gen_for(). 3 different basic blocks are defined (condition, body, final), and the condition block is pushed on the continue stack.
If there is an iterator expression, a 4th block "iterator" should be defined and pushed on the continue stack instead of the condition block. This block gets control at the end of the body block or if the body block contains a continue.

windelbouwman, 2020-06-10:
Addressed in d024ae96e648d21c2b373ebce383e81d06bd74cd thanks to your good explanation!

tstreiff, 2020-06-10:
The fix is OK!

closed 2020-06-10 by windelbouwman