windelbouwman/ppci #90: Polyhedral optimization

windelbouwman, 2020-06-15:
This looks cool:

https://polly.llvm.org/performance.html

How would we integrate this?

pfalcon, 2020-06-15:
What's so cool about? Polyhedral-model optimization looks like "corporate bloat" of compiler optimizations. Loop parallelization for multiprocessors and the like - what can be more boring? ;-) People don't build such CPU on kitchen tables.

I recently looked for what loop optimizations PPCI does at all, and saw nothing. I'd suggest to start with classic loop optimizations like loop invariant code motions and induction variable optimization.

pfalcon, 2020-10-23:
Related: https://github.com/inducer/loopy . "A code generator for array-based code on CPUs and GPUs". Pure-Python. It doesn't mention "polyhedral optimization" specifically, but does some loop munging associated with it (like tiling, etc.).