windelbouwman/ppci #96: ppci-cc: floating point constants with an exponent but no decimal point are rejected

tstreiff, 2020-06-23:
Current version rejects constants such as 12e3. 

>  f = 12e3;
>          ^ Expected ";", got "ID"

but constants like 1.2e4 are accepted.
Since there is no decimal point the lexer takes "12" for an integer constant, and assumes that "e3" is part of an following ID.

The C standard specifies that a decimal floating point constant may be:

> [digit-sequence] . digit-sequence [exponent-part]
> digit-sequence . [exponent-part]
> digit-sequence exponent-part

windelbouwman, 2020-07-12:
This should be fixed by commit 0e9e024a66df4dc5182f241056464f6328a36bae.

@tstreiff could you verify and close if this is working fine now?

tstreiff, 2020-07-21:
I have tested a few cases.
12e3 is now accepted correctly.

However, the following constants are still rejected by the parser (they are mostly corner cases):
    0.7e0
    0.
    0e1
    0e-0
The following constant makes the IR generator crash:
    78E-1
The following constant is accepted silently although it should be flagged as too big:
    9.0E122334
Last, any floating point constants with a suffix is rejected by the parser (f, F, l, L)
    1.2e10f

I attach below a test file (issue96.c) covering most cases, including the failed ones listed above.
[issue96.c.txt](https://github.com/windelbouwman/ppci/files/4952102/issue96.c.txt)