windelbouwman/ppci #107: ppci-cc: initializing a global enum variable is not possible

tstreiff, 2020-07-08:
enum E { ONE, TWO, THREE };
    
    enum E g0;          // compiles
    enum E g1 = ONE;    // compilation exception: NotImplementedError: Enum-type
    enum E g2 = 1;      // compilation exception: NotImplementedError: Enum-type                                      

    int main() {
        enum E l1 = ONE;  // compiles
        enum E l2 = 2;    // compiles
        return 0;
    }

The compiler raises an exception at any global enum type variable initialization.
Local initializations work (l1, l2)

windelbouwman, 2020-07-08:
This should be fixed now. I added a fix + the example you provided. Thanks for reporting!

tstreiff, 2020-07-21:
Good! Test passed, I close the issue.

closed 2020-07-21 by tstreiff