windelbouwman/ppci #113: Implementation of sbrk

darleybarreto, 2020-08-30:
In order to implement dynamic allocated str (#91) and more, today I worked on bringing malloc support. I am almost there (I altered the necessary python code for that), the only thing remaining is `sbrk` function, following [this](https://mcuoneclipse.com/2014/03/16/freertos-malloc-and-sp-check-with-gnu-tools/) implementation I would need to link the information about the end of the free memory on the host. What would be the best way of doing that?

windelbouwman, 2020-08-31:
I'm not sure how to do that, but you could use a linker script to do so? From the memory map you could define a symbol and use that in an assembly level implementation of sbrk.

I'm not sure what you want to achieve, do you want to implement your own sbrk function?

darleybarreto, 2020-08-31:
I want to implement malloc, so we could allocate arbitrary objects inside ppci (for unix based systems with brk similar syscalls). The "easy" way would be to implement brk syscall and sbrk wrapper, then malloc using sbrk.

windelbouwman, 2020-08-31:
If you want to implement syscalls, I would checkout this sample: https://github.com/windelbouwman/ppci/blob/master/examples/linux64/hello-make/hello.c

darleybarreto, 2020-08-31:
Oh yeah, I have seen this one, and I altered the implementation of [do_inline_asm](https://github.com/windelbouwman/ppci/blob/master/ppci/codegen/irdag.py#L361), so one can get the return of a syscall in a variable like this:
```c
long syscall(long nr, long a, long b, long c)
{   
    long ret;

    asm(
        "mov rax, %0 \n"
        "mov rdi, %1 \n"
        "mov rsi, %2 \n"
        "mov rdx, %3 \n"
        "syscall \n"
        : "=r" (ret)
        : "r" (nr), "r" (a), "r" (b), "r" (c)
        : "rax", "rdi", "rsi", "rdx"
    );

    return ret;
}
```
In `ret` we would have the break point given by the system, so sbrk would work with this value to give the start memory address when called inside a malloc implementation.

darleybarreto, 2020-09-06:
I worked out the syscalls and made a port of a simple malloc. So now we can do something like this
```c
#include <stdio.h>
#include <stdlib.h>

int main(){
   int *int_vec = (int*)malloc(10*sizeof(int));
   for (int i =0; i < 10; i++){
      int_vec[i] = i*i + 2;
   }
   
   for (int i =0; i < 10; i++){
      printf("%d",int_vec[i]);
   }

   free(int_vec);

   exit(0);
}
```

windelbouwman, 2020-09-07:
Cool! Nicely done!!

darleybarreto, 2020-09-07:
I also took the liberty to change some file structures inside `librt/libc`, I guess you will check if it is ok when reviewing the PR. I've never submitted a PR before, should I do something w.r.t CI/CD?

windelbouwman, 2020-09-07:
Sure, feel free to create a PR, lets see what happens :). No need to do any CI/CD. If you want, you can run tests locally with tox / pytest as described in the docs, but I guess this is done as well with CI/CD on github.

darleybarreto, 2020-09-13:
Closed by #114

darleybarreto, 2020-09-13:
Closed by #114

closed 2020-09-13 by darleybarreto