windelbouwman/ppci #122: Translating to ppci assembly

darleybarreto, 2021-01-30:
Hi, I am not a system programmer so I have really limited skills on this subject, specially when comes to assembly. I am trying to make x86_64 musl `longjmp` work on ppci, [this](https://github.com/bminor/musl/blob/master/src/setjmp/x86_64/longjmp.s) is the implementation:
```asm
.global longjmp
longjmp:
	xor %eax,%eax
	cmp $1,%esi             /* CF = val ? 0 : 1 */
	adc %esi,%eax           /* eax = val + !val */
	mov (%rdi),%rbx         /* rdi is the jmp_buf, restore regs from it */
	mov 8(%rdi),%rbp
	mov 16(%rdi),%r12
	mov 24(%rdi),%r13
	mov 32(%rdi),%r14
	mov 40(%rdi),%r15
	mov 48(%rdi),%rsp
	jmp *56(%rdi)           /* goto saved address without altering rsp */
```
Is there a guide to the asm that ppci reads? I know some things, but from simple examples. The above snippet should be something like the following:
```asm
global longjmp
longjmp:
	xor eax,eax
	cmp #1,esi             ; CF = val ? 0 : 1
	adc esi,eax           ; eax = val + !val
	mov (rdi),rbx         ; rdi is the jmp_buf, restore regs from it
	mov 8(rdi),rbp
	mov 16(rdi),r12
	mov 24(rdi),r13
	mov 32(rdi),r14
	mov 40(rdi),r15
	mov 48(rdi),rsp
	jmp *56(rdi)           ; goto saved address without altering rsp
```

pfalcon, 2021-01-30:
There're 2 main assembly syntaxes for x86 - "Intel" and "AT&T". Clearly, PPCI follows "Intel" syntax, you can google up differences between the two.

pfalcon, 2021-01-30:
In regard to classic "Intel" syntax, PPCI clearly lacks (in)famous `byte ptr` and friends (PPCI never generates memory/immediate instructions?) and uses `[rbp, 8]` instead of `[rbp + 8]` for indirect memory access.

Don't forget `-S` switch to `ppci-cc`.

windelbouwman, 2021-01-31:
>  I am not a system programmer

Not yet, you will be soon! :stuck_out_tongue_closed_eyes: 

The assembly syntax is sort of intel style, as mentioned. It would be a cool addition to be able to support more than one syntax, but this would probably be a big rework of the current system.

darleybarreto, 2021-01-31:
So should it be something like this?
```asm
global longjmp
longjmp:
	xor eax, eax
	cmp esi, 1             ; CF = val ? 0 : 1 
	adc eax, esi           ; eax = val + !val   
	mov rbx, [rdi]         ; rdi is the jmp_buf, restore regs from it 
	mov rbp, [rdi,8]
	mov r12, [rdi,16]
	mov r13, [rdi,24]
	mov r14, [rdi,32]
	mov r15, [rdi,40]
	mov rsp, [rdi,48]
	jmp [rdi,56]           ; goto saved address without altering rsp 
```

darleybarreto, 2021-02-03:
So, I defined the `jmp_buf` as:
```c
typedef unsigned long jmp_buf[8];
```
Following [this](https://gitlab.redox-os.org/redox-os/relibc/-/blob/master/src/header/setjmp/impl/x86_64/setjmp.s) implementation, `setjmp` is 
```asm
global setjmp
setjmp:
	mov [rdi], rbx          ; rdi is jmp_buf, move registers onto it 
	mov [rdi,8], rbp
	mov [rdi,16], r12
	mov [rdi,24], r13
	mov [rdi,32], r14
	mov [rdi,40], r15
	lea rdx, [rsp,8]        ; this is our rsp WITHOUT current ret addr 
	mov [rdi,48], rdx
	mov rdx, [rsp]          ; save return addr ptr for new rip 
	mov [rdi,56], rdx
	xor rax,rax             ; always return 0 
	ret
```
and [longjmp](https://gitlab.redox-os.org/redox-os/relibc/-/blob/master/src/header/setjmp/impl/x86_64/longjmp.s):
```asm
global longjmp
longjmp:
	mov rax, rsi            ; val will be longjmp return 
	test rax,rax
	jne __longjmp
	inc rax                 ; if val==0, val=1 per longjmp semantics 
__longjmp:
	mov rbx, [rdi]          ; rdi is the jmp_buf, restore regs from it 
	mov rbp, [rdi,8]
	mov r12, [rdi,16]
	mov r13, [rdi,24]
	mov r14, [rdi,32]
	mov r15, [rdi,40]
	mov rsp, [rdi,48]       ; this ends up being the stack pointer 
	mov rsp, rdx
	mov rdx, [rdi,56]       ; this is the instruction pointer 
	jmp [rdx]               ; goto saved address without altering rsp
```
`setjmp` seems to work fine, but `longjmp` is seg faulting in this code:
```c
#include <stdlib.h>
#include <stdio.h>
#include <setjmp.h>

static jmp_buf jmp_buffer;

enum { OK, ERR_DIV_BY_ZERO };

int div(int a, int b) {
    if(b==0) {
        longjmp(jmp_buffer, ERR_DIV_BY_ZERO);
    }

    return a/b;
}

void main() {
    int result, error;
    error = setjmp(jmp_buffer);

    if(OK==error) { /* try */
        result = div(1, 0);
    }
    else { /* catch */
        printf("Error in div().");
        exit(1);
    }

    printf("Result: %d\n", result);
    exit(0);
}
```
The difference between the original implementation is that I use `jne __longjmp` instead of `jnz __longjmp`, which should be the [same](http://unixwiz.net/techtips/x86-jumps.html).

darleybarreto, 2021-02-05:
Replacing the previous `longjmp` asm with the following from newlibc does the trick
```asm
global longjmp
longjmp:
	mov rax, rsi            ; val will be longjmp return 
	mov rbp, [rdi,8]
	mov rsp, [rdi,48]       ; this ends up being the stack pointer 
	mov rbx, [rdi,56]
	push rbx
	mov rbx, [rdi,0]          ; rdi is the jmp_buf, restore regs from it 
	mov r12, [rdi,16]
	mov r13, [rdi,24]
	mov r14, [rdi,32]
	mov r15, [rdi,40]
	ret
```

closed 2021-02-05 by darleybarreto

windelbouwman, 2021-02-06:
Well played, this is a tricky thing involving stack layout, so this is depending upon register saved variables. Glad you got it to function!